FORTH-ModelBasedTracker/HandTracker #2: undefined Symbol

MarkusEich, 2015-11-16:
Running the script results in and error

Traceback (most recent call last):
  File "src/SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: /home/eich/devel/MBV_PythonAPI_Linux_1.1/libs/libCore.so: undefined symbol: _ZN2cv9ExceptionC1EiRKSsS2_S2_

System: CUDA  7.5, NVIDIA 352, Ubuntu 15.10

padeler, 2015-11-17:
Hello, 

From the error message it looks like the cv::Exception is undefined. Please make sure that all opencv packages are installed on your system. (libopencv-dev, python-opencv)

severin-lemaignan, 2015-11-17:
Hello, same issue here on Ubuntu 15.10. OpenCV 2.4.9 and all Python bindings are installed.

Here the output of `ldd` on `libCore`:

```
$ ldd libs/libCore.so 
	linux-vdso.so.1 =>  (0x00007fffb2117000)
	libOpenMeshCore.so.3.1 => not found
	libcudart.so.7.5 => not found
	libboost_system.so.1.57.0 => not found
	libboost_filesystem.so.1.57.0 => not found
	libboost_thread.so.1.57.0 => not found
	libboost_serialization.so.1.57.0 => not found
	libboost_program_options.so.1.57.0 => not found
	libboost_log.so.1.57.0 => not found
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ffad74b4000)
	libtbb.so.2 => /usr/lib/libtbb.so.2 (0x00007ffad7275000)
	libopencv_core.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 (0x00007ffad6e4a000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ffad6ac8000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ffad67bf000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ffad65a8000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ffad61de000)
	/lib64/ld-linux-x86-64.so.2 (0x0000560399802000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ffad5fd9000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ffad5dbf000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007ffad5bb7000)
	libGL.so.1 => /usr/lib/nvidia-352/libGL.so.1 (0x00007ffad5886000)
	libnvidia-tls.so.352.41 => /usr/lib/nvidia-352/tls/libnvidia-tls.so.352.41 (0x00007ffad5683000)
	libnvidia-glcore.so.352.41 => /usr/lib/nvidia-352/libnvidia-glcore.so.352.41 (0x00007ffad2bf0000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007ffad28b6000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007ffad26a4000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007ffad2482000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007ffad227e000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007ffad2078000)
```

Surely enough, OpenCV is correctly found. So I'm not sure why `cv::Exception` is not found.

There are also other issues here (beyond `libcuda` not found -- I've not installed it yet):
- It seems that the library was compiled against `boost 1.57` (Ubuntu 15.10 ships with `boost-1.58`).
- I'm not sure what `libOpenMeshCore` is, but it seems it is not packaged in ubuntu.

padeler, 2015-11-17:
Hello, 

The libraries you are missing are inside the binaries package in the "libs" folder. You need to set the MBV_LIBS environment variable as explained in the Readme.md. Alternatively you can set the LD_LIBRARY_PATH and PYTHONPATH manually. Check the runme.sh for details.

severin-lemaignan, 2015-11-17:
Well, if I got the error with the missing symbol in `libCore` in the first place, it is because `libCore` was found, ie, `MBV_LIBS` was correclty exported ;-)

severin-lemaignan, 2015-11-17:
Hum, I see, you manipulate the `LD_LIBRARY_PATH` in `runme.sh`. Alright. I'll conduct further tests, then.

severin-lemaignan, 2015-11-17:
Could you tell us which version of OpenCV you have been using to compile the library?

padeler, 2015-11-17:
The libraries where compiled with Ubuntu 14.04 which comes with OpenCV 2.4.8

severin-lemaignan, 2015-11-17:
Alright, I think I've figured out the reason: on Ubuntu 15.10, the default compiler is GCC 5, which effectively changes the ABI of `std::string` ([see here](http://developerblog.redhat.com/2015/02/05/gcc5-and-the-c11-abi/) for instance).

One option is to recompile a local OpenCV with gcc < 5, or to recompile OpenCV with gcc 5 and the additional flag `-D_GLIBCXX_USE_CXX11_ABI=0`.

You may or may not want to provide binaries compiled (well, linked) on a system using gcc 5 as default compiler to avoid the issue (or alternatively, ship the required opencv libraries as well).

padeler, 2015-11-17:
Thanks for the find Severin,

To keep things simple we decided to only offer binaries for the LTS release of Ubuntu and tried to avoid packaging libs that are provided by the system. Obviously this leads to issues. Maybe in the next binaries update we will also include the OpenCV libs.

MarkusEich, 2015-11-17:
Thank you for the update. Maybe it is easier to have several ubuntu versions for the compiled libs? At least for the LTS and for the current version? Would keep the libs shipped with the tracker to a minimum. 

Thanks 

Markus

MarkusEich, 2015-11-18:
Confirmed the work around for the issue. After downloading OpenCV 2.4 and compiling it using 

# export CC=/usr/bin/gcc-4.9
# export CXX=/usr/bin/g++-4.9
# cmake .. -D_GLIBCXX_USE_CXX11_ABI=0 -DCUDA_HOST_COMPILER=/usr/bin/gcc-4.9 

The tracker runs smoothly under Ubuntu 15.10, CUDA 7.5

Thnx guys

padeler, 2015-11-18:
Thanks for the confirmation Markus. I am closing this issue.

closed 2015-11-18 by padeler