FORTH-ModelBasedTracker/HandTracker #4: Error at startup

severin-lemaignan, 2015-11-17:
When running the tracker on Linux (Ubuntu 15.10), I get the following errer:

```
$ ./runme.sh 
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS /home/slemaignan/applis/MBV_PythonAPI_Linux_1.1/
Setting LD_LIBRARY_PATH and PYTHONPATH
Running the Single Hand Tracker script...
Creating Renderer...
[2015-11-17 13:14:16.524800] [0x000007fa9541c170] [warning] Read options from environment : unrecognised option 'libs'
[2015-11-17 13:14:16.524891] [0x000007fa9541c170] [warning] Read options from environment : unrecognised option 'libs'
Traceback (most recent call last):
  File "src/SingleHandTracking.py", line 33, in <module>
    renderer = Rendering.RendererOGLCudaExposed.get()
RuntimeError: /home/padeler/work/mbv/release_libs/src/mbv/Rendering/RendererOGL/RendererOGL.cpp(1142): Throw in function MBV::Rendering::RendererOGLCudaExposed::RendererOGLCudaExposed(MBV::Core::UInt, MBV::Core::UInt, MBV::Core::UInt, const String&, const String&)
Dynamic exception type: boost::exception_detail::clone_impl<MBV::Core::Exception>
[MBV::Core::Message*] = unknown error
```

Any idea?

padeler, 2015-11-17:
The exception is raised when the Renderer tries to initialize the CUDA device. What is your GPU and driver version? If you are running on a system with dual GPUs you should make sure the process is started with optirun using the NVidia gpu.

severin-lemaignan, 2015-11-17:
```
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: Quadro K2200/PCIe/SSE2
OpenGL core profile version string: 4.4.0 NVIDIA 352.41
OpenGL core profile shading language version string: 4.40 NVIDIA via Cg compiler
```

nkyriazis, 2015-11-17:
Is this the display GPU? Have you enabled e.g. a M/B onboard graphics adapter or any other device that acts as the first (0) display adapter?

severin-lemaignan, 2015-11-17:
Yes, it is DISPLAY 0. I'm going to update to 352.63 (the latest version) to see if it solves the issue.

nkyriazis, 2015-11-17:
Can you confirm that you can successfully execute other cuda-enabled applications? E.g. NVidia cuda samples.

severin-lemaignan, 2015-11-17:
Alright: the tentative update to NVIDIA 352.63 was a total failure (had to physically remove my graphic card to boot again...), so I'll keep 352.41 for now.

After re-installing the CUDA 7.5 SDK (following [these instructions on Ubuntu 15.10](http://askubuntu.com/questions/693145/installing-cuda-7-5-toolkit-on-ubuntu-15-10)), the hand tracker successfully starts. Cool! Closing the issue, but you may want to make it clear that the user must install the CUDA SDK *even though* `libcudart.so` is provided with the libraries.

closed 2015-11-17 by severin-lemaignan