FORTH-ModelBasedTracker/HandTracker #6: Live capture error at startup

gsaponaro, 2015-11-17:
Hello, my system is Ubuntu 14.04 LTS, NVIDIA driver 352.63, CUDA 7.5.

When I run the tracker with the provided oni recording, it works.

Now I'd like to try the live capture from a sensor (Asus Xtion Pro Live). I set the path to the empty string (```oniPath = ''```) but I obtain the following error.

```
$ ./runme.sh 
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS /usr/local/src/robot/MBV_PythonAPI_Linux_1.1
Setting LD_LIBRARY_PATH and PYTHONPATH
Running the Single Hand Tracker script...
Creating Renderer...
[2015-11-17 18:05:42.675324] [0x000007f3792a8d74] [warning] Read options from environment : unrecognised option 'libs'
[2015-11-17 18:05:42.675436] [0x000007f3792a8d74] [warning] Read options from environment : unrecognised option 'libs'
  Warning! Material file '/usr/local/src/robot/HandTracker/media/sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file '/usr/local/src/robot/HandTracker/media/cylinder.mtl' not found!
Warning! Material '' not defined in material file.
Variances:  [10.0, 10.0, 10.0, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]
Low Bounds:  [-2000.0, -2000.0, -2000.0, -1.0, -1.0, -1.0, -1.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 2.5, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0]
High Bounds:  [2000.0, 2000.0, 2000.0, 1.0, 1.0, 1.0, 1.0, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.0, 3.140000104904175, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874]
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
Starting Grabber...
[2015-11-17 18:05:43.180230] [0x000007f3792a8d74] [error]   XN node not present. One or more of the following nodes could not be enumerated:

[2015-11-17 18:05:43.180337] [0x000007f3792a8d74] [error]   Failed to initialize openni_acquisition plugin. /home/padeler/work/mbv/release_libs/src/mbv/Acquisition/OpenNIGrabber/OpenNIGrabber.cpp(166): Throw in function void MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: boost::exception_detail::clone_impl<MBV::Core::Exception>
[MBV::Core::Message*] = OpenNI error. Node not present

Entering main Loop.
[2015-11-17 18:05:43.180441] [0x000007f3792a8d74] [error]   Xn function Failed: Input pointer is null!
```

First, it looks like some material files are not found in the media directory. In fact I don't see any .mtl there, only .obj and .xml.

Then, the OpenNIGrabber that is being called is from a /home/padeler/... path instead of my local path.

Any suggestion? Many thanks.

padeler, 2015-11-17:
Hello, 
The material file is missing but it is not used by the tracker (the warning is output of the mesh loader).

The issue you are having is related to the sensor which is not setup properly. 
Make sure the driver is installed properly (openni 1.5 and sensor driver). You can check if your camera is working by running the NiViewer application that comes with OpenNI.

padeler, 2015-11-19:
I am closing this since this is a camera/driver related issue.

closed 2015-11-19 by padeler

gsaponaro, 2015-11-25:
Thanks and sorry for the late response. For the record, I solved the issue by disabling the internal USB3.0 controller in the BIOS.

HeGaoYuan, 2016-03-27:
Hi, I also meet the same problem like you.
You said you have solved the issue by disabling the usb3.0. 
Actually, I connected my kinect1 with my computer in the usb2.0 port. So, I think it is not about the usb 3.0

gsaponaro, 2016-03-28:
Hi @HeGaoYuan, can you successfully run NiViewer from the OpenNI samples?