FORTH-ModelBasedTracker/HandTracker #7: Very slow startup even with increasing cache size

MarkusEich, 2015-11-18:
After starting the runme.sh script it takes EVERY  time around 3! minutes until the window appears. 
Even after setting the CUDA_CACHE_MAXSIZE=2147483648, as recommended.

Is there any way to speed up the process?

System used: Ubuntu 15.10, CUDA 7.5, NVIDIA GTX860M (4GB RAM), 16GB RAM, Intel I7 CPU

padeler, 2015-11-18:
The slow startup speed is due to the JIT compilation of the CUDA code. Since the generated binaries are bigger than the default cache, the workaround is to increase the max cuda cache size (this is what nvidia also suggests). Maybe your cache size is set somewhere else? Try setting it in your `.bashrc` file

MarkusEich, 2015-11-18:
Checked setting with env. Gives a chache size of CUDA_CACHE_MAXSIZE=2147483648. Even tried with double size (4GB). The folder ~/.nv is there and readable/writable. 

I read that JIT compilation takes place if the host architecture is different from the compile architecture. Maybe if all features are compiled for all architectures (i.e. sm_10,sm_13,sm_20,sm_30,sm_35), the JIT will not be necessary  (but I might be wrong).

Anyone else having the slow start-up issue?

MarkusEich, 2015-11-18:
Solved. I had to set explicitly export CUDA_CACHE_DISABLE=0. Thought it would have been enabled by default. First run now takes 2 minutes, second run 2 seconds.

closed 2015-11-19 by nkyriazis