FORTH-ModelBasedTracker/HandTracker #8: HandTracker with Kinect v2

Sahbaap, 2015-11-18:
Hi,

I followed the instructions you provided n could run runme.bat from the command line which shows a person whose right hand is being tracked. I am just wondering if there is a sample code of your software that works with Kinect v2 in which I can test it. 

Thanks!

nkyriazis, 2015-11-19:
I infer that you're using the Windows version.

**Note:** The Kinect2 acquisition is only available through the MBV libraries for the Windows OS, where official support is provided. This does not mean that the hand tracker cannot be used with Kinect2 input. It only means that such input might have to be acquired by means different than MBV (non-Windows).

I'll commit an example using Kinect2 and let you know.

nkyriazis, 2015-11-19:
Please have a look at the [Kinect2 branch](https://github.com/FORTH-ModelBasedTracker/HandTracker/tree/Kinect2Input). Changes are made in the python script to use live Kinect2 input in low resolution (register rgb to depth) or high resolution (register depth to rgb).

Let me know if it works for you.

Sahbaap, 2015-11-19:
Perfect thanks, very nice! Yeah I could use your library and track my hand with a pretty good accuracy. I have two more questions. 

The first question is that how fast my hands can be moved that the handtracker does not the get lost? cuz I noticed if I move my hands fast it does get lost. 

And my second question is that is there any way that we can skip the initialization part of the code? Cuz the application that I wanna use your library is for stroke rehabilitation and I imagine that the initialization step would be somehow annoying for stroke patients.

nkyriazis, 2015-11-20:
Happy to hear it works for you. I'm closing these issue

* Hand motion speed: See #9 
* Initialization: See #10

closed 2015-11-20 by nkyriazis