FORTH-ModelBasedTracker/HandTracker #11: gcc 5 ABI

MarkusEich, 2015-11-23:
I am running the Handtracker on Ubuntu 15.10 with gcc 5.2. The issue is that opencv on the system has to be compiled with gcc <5.0 in order to be working with the hand tracker libs. Problem is the new ABI introduced with gcc 5.0

This is fine BUT if ROS comes into play things get worst. ROS has to be compiled with gcc< 5.0 as well in order to be compatible with the hand tracker. This fails because the system BOOST is not compatible with gcc < 5.0 and so on, resulting in building Boost as well with gcc< 5.0 and so on. Would it be possible to have two versions of the hand tracker binaries compiled? One with gcc<5.0 and one with gcc>5.0? This would solve many issues with Ubuntu > 14.04 (which is almost two years old now).

Thanx

padeler, 2015-11-23:
Hello,

As far as I understand, since you have opencv compiled and working with the tracker, you just need to copy the binaries of opencv in the libs folder of the tracker (not a system wide install). The rest of the system (including ROS) will use the gcc-5 binaries (newer opencv etc). So the solution is to just use the binaries of opencv with the tracker.

We decided to provide binaries only for the LTS releases of ubuntu to keep the release overhead low. The next LTS will be 16.04.

MarkusEich, 2015-11-23:
Thanx for the answer. 

The problem seems to arise because I am implementing a node which receives images via ROS and processes it via the hand tracker. So this process needs to be linked to the 4.9 libs of open CV and ROS libs compiled with gcc 5.0. Patently you cannot mix the libs. This results in undefined references in one or the other. 

Cheers 
Markus

MarkusEich, 2015-11-24:
I think it should be sufficient to provide ABI C++ 11 compatibility for the libCore.so of the hand tracker lib. I think the lib should work than with LTS 14.04 as well as 15.10. Could we maybe try this? Some information is given in http://developerblog.redhat.com/2015/02/05/gcc5-and-the-c11-abi/

closed 2017-09-11 by padeler