FORTH-ModelBasedTracker/HandTracker #12: Support for manually installed external Linux dependencies

gsaponaro, 2015-11-25:
It would be nice if the external dependencies in Linux (mainly OpenCV, if possible also OpenNI1) were supported in the case where they were compiled and installed manually by the user (ideally in a location corresponding to an enviroment variable), rather than with the binary Ubuntu packages.

For example, I normally use a recent version of OpenCV (newer than 2.4.8 which is the current ```libopencv-dev```), compiled with some custom flags, to run other vision projects. I would like to use your hand tracker with that version if possible, rather than having to install another OpenCV system-wide, which can override the dependencies and builds in my "other" projects. Not super critical or annoying, but it would be a nice addition for the future. Cheers.

padeler, 2015-11-25:
Good point. The idea was to use as many of the system provided libraries as possible in order to keep the package size small. It turns out it is not the most convenient choice in some cases.

closed 2017-09-11 by padeler

gsaponaro, 2017-09-11:
Hi @padeler, out of curiosity, are there any plans for supporting this? I understand that time and resources are limited for you guys. I could help on this if there was a CMakeLists.txt, having some familiarity with these library issues in CMake.

Thanks as usual for your help and for the hand tracker software.

padeler, 2017-09-11:
There has been major changes to the library over the past year. We plan a new release soon that will also solve issues like this one.