FORTH-ModelBasedTracker/HandTracker #13: Using real sensor: Xn function Failed: File not found!

MarkusEich, 2015-11-26:
I want to run the hand tracker with a kinect, so I changed line 92  to

acq = Acquisition.OpenNIGrabber(True, True, '', oniPath, True)
in order to run with the Kinect attached. Openni is installed properly. Running the script gives an error:

Starting Grabber...
[2015-11-26 15:16:03.732424] [0x000007fe109fae70] [error]   Xn function Failed: File not found!
[2015-11-26 15:16:03.732478] [0x000007fe109fae70] [error]   Failed to initialize openni_acquisition plugin. /home/padeler/work/mbv/release_libs/src/mbv/Acquisition/OpenNIGrabber/OpenNIGrabber.cpp(171): Throw in function void MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: boost::exception_detail::clone_impl<MBV::Core::Exception>
[MBV::Core::Message*] = OpenNI error. Function Failed

Entering main Loop.
[2015-11-26 15:16:03.732538] [0x000007fe109fae70] [error]   Xn function Failed: Input pointer is null!

Any ideas?

padeler, 2015-11-26:
Did you press 's'? Does the script work with th .oni file ? Give some more details on the output of the tracker.

MarkusEich, 2015-11-26:
Yes the tracker is working with the oni file.  What is done with the s key? Can check tomorrow. (hope s is not the start button :-)

padeler, 2015-11-26:
's' starts/stops the tracking. Check the comments inside the script.

closed 2015-11-26 by padeler

MarkusEich, 2015-11-26:
Will check tomorrow. If it works please remove this comment. This is embarrassing :-)