FORTH-ModelBasedTracker/HandTracker #14: OpenGL error in running runme.bat

ma581, 2015-11-27:
Hi please could you help me with the following error: 

ERROR MESSAGE:
C:\Users\CIT-guest\Documents\GitHub\HandTracker>runme.bat
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS C:\Users\CIT-guest\Documents\GitHub\MBV_PythonAPI_Win_1.1
Setting PATH and PYTHONPATH
"Running the Single Hand Tracker script..."
Creating Renderer...
[2015-11-26 18:11:50.513381] [0x000001cf] [warning] Read options from environment : unrecognised option 'libs'
[2015-11-26 18:11:50.513381] [0x000001cf] [warning] Read options from environment : unrecognised option 'libs'
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 33, in <module>
    renderer = Rendering.RendererOGLCudaExposed.get()
RuntimeError: E:\mbv\Rendering\RendererOGL\OpenGLHelpers.cpp(158): Throw in function void __cdecl OGL::ThrowOnCompileLinkError(unsigned int,enum OGL::CompLinkMode,const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = error while compiling file :
ERROR: 0:2: 'flat' : syntax error syntax error

******** ******** ******** 
 MY SETUP
******** ******** ******** 
HARDWARE
- Intel Core i7
- 8 GB of RAM
- GPU: Nvidia GeForce GT 555M
- CUDA: 7.5.18
- OpenGL 4.5 

SOFTWARE
OS
- 64bit Windows 10 

Environment
- Python 2.7 64bit

Drivers
- Latest CUDA driver
- OpenNI driver (OpenNI 1.x SDK for Windows 8 64bit and OpenNI 1.x sensor driver for 8 Windows 64bit)
- Kinect 2 driver (Kinect 2 SDK for Windows 8 64bit and newer)

nkyriazis, 2015-11-27:
The error is probably OpenGL/driver related. Let me try and explain.

The tracker utilizes a renderer, which in turn employs a vertex shader, a geometry shader and a fragment shader. The vertex and geometry shaders output ['flat' values](https://www.opengl.org/wiki/Type_Qualifier_(GLSL)#Interpolation_qualifiers), i.e. values that are not interpolated in the fragment shader. This is implemented by attributing inputs/outputs with the 'flat' qualifier. All utilized features are available in OpenGL 3.3. Every shader carries a <tt>#version 330 core</tt> preamble.

I would initially focus on the fact that the GPU is a mobile version. It is not uncommon that ports of desktop drivers to mobile variants introduce feature support issues, in contrast to the alleged OpenGL support.

ma581, 2015-12-03:
I was able to get runme.bat running on a different computer with an Nvidia GeForce GTX 860M :)

closed 2015-12-03 by nkyriazis