FORTH-ModelBasedTracker/HandTracker #16: write down articulation information to file

koutaka, 2015-12-17:
I wanna write down articulation information to file in 3D or 2D dimensional image, but I didn't know how do it.
How should I write down it to file?

nkyriazis, 2015-12-18:
The question is not clear enough. I'll try to explain as much as possible. Consider the following pieces of code in conjunction with the provided hand tracking python script.

During the tracking loop a variable <tt>currentHandPose</tt> is updated. This is the 27-D parametric representation of a hand 3D configuration (pose + articulation). You can convert it to a <tt>numpy.ndarray</tt> and process/save it any way you like:

```python
# Convert to numpy
currentHandPose = numpy.array(currentHandPose)
```

The 27 values are converted to an assembly of 3D geometries through a process we generally refer to as "decoding". To get detailed 3D information for joint locations you need to decode <tt>currentHandPose</tt>. To get 2D locations you need to project using the camera calibration.

```python
# get camera calibration
frustum = clbDepth.camera
proj = frustum.Graphics_getProjectionTransform()
view = frustum.Graphics_getViewTransform()
viewport = frustum.Graphics_getViewportTransform(640, 480)

# break down 27-D solution to assembly of 3D transforms
decoding = ht.decoder.quickDecode(currentHandPose)
zero = Core.Vector4(0, 0, 0, 1)
for d in dec.values():
    for m in d.matrices:
        # The points computed below have members x, y, z and w
        # Don't neglect w, as some transforms are projective
        # 3D point in homogeneous coordinates
        pt3D = view * m * zero
        # 2D projected point in homogeneous coordinates
        pt2D = viewport * proj * view * m * zero
```

ma581, 2015-12-21:
I am using the SingleHandTracking.py fom the branch Kinect 2.  

1. Using the conversion to an array for currentHandPose gives me a 9x3 Matrix :+1: 
2. However, I have issues in "decoding" using the lines of code you detailed:

### Error Message ###
Entering main Loop.
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 232, in <module>
    for d in dec.values():
AttributeError: 'module' object has no attribute 'values'

koutaka, 2015-12-21:
Hi, ma581. I solved this problem before.
If you replace "dec" with "decoding", your program will move. 
A value of "pt2D" is bigger than 2D image, however, value is odd.
I wanna figure it out. Could you tell me how to solve if you will be clear and I use Kinect v2, too.

ma581, 2015-12-21:
Thanks @koutaka !

nkyriazis, 2015-12-22:
Corrected the typo.

Both the 3D and 2D points are in 3D homogeneous space (4D). Don't forget to incorporate w. Some parts of the hand might actually lie outside the image, so this is not alarming in general. The clipping test is
```
-w <= (x,y,z) <= w
```

ma581, 2016-02-07:
I am trying to get rotation information of the finger joints. 

1. I am able to get positions of the finger joints in homogeneous coordinates using pt3D.x, pt3D.y etc
2. How would I get rotations of these same finger joints (within the same coordinate system)? Are the quaternions in the decoding object?

nkyriazis, 2016-02-08:
The finger angles are encoded in relative angles in the 3D hand pose parameterization. More specifically, the 3D hand pose comprises:

palm position (3D) + palm orientation (Quaternion) + 5 X (finger pitch, finger yaw, finger joint 1, finger joint 2)

The kinematics are output, through decoding, into absolute world space transforms. You could either consult the original representation of reconstruct relative angles from the absolute transforms.

closed 2016-02-17 by nkyriazis

Freshy123, 2016-05-05:
Good Afternoon,

@ma581 I see that you are able to obtain the position of the finger joints in a 3D space, would you be able to send me over a version of your code so that I may see what you have done to achieve this? I am looking for the positions of the finger tips on the model. Also if you are willing would you be able to give a brief explanation on how you obtained the data given your read our data?

feel free to contact me via the information below.

Very Respectfully,

Haptic Tracking group 
Rochester Institute of Technology
dpc9190@rit.edu
716-421-4181

kheito, 2017-04-10:
May I ask, where do the values of the 4D rotation(quaternion) are coming from? I mean is there a specific point location for the x,y,z and most specially, w? Thank you in advance

padeler, 2017-04-10:
The quaternion represents the global rotation of the hand model (the root of the model is on the wrist).
You can check the related publications for more info (i.e. http://users.ics.forth.gr/~argyros/mypapers/2011_09_bmvc_kinect_hand_tracking.pdf)

kheito, 2017-06-15:
I ust wan't to know one thing. Do the detection or value of 4d rotation of the wrist movement (x,y,z,w) will be affected or be inaccurate if the fingers are all close? Hope you'll find time to answer. Thank you very very much.

padeler, 2017-06-19:
Yes it can be affected, if the fingers are closed (open palm, fingers joint) there is an ambiguity in the appearance of the hand in the depth map and the location and rotation of the wrist can drift a little. That said, if the hand changes pose to a less ambiguous pose, the estimation of the wrist pose will recover.