FORTH-ModelBasedTracker/HandTracker #22: Xn function Failed: Input pointer is null!

shomedas, 2016-02-16:
I changed the onipath to empty string to perform live capture from an existing sensor 
oniPath = ''
when i run .runme.sh i fails to find the sensor. The sensor however is working properly, I checked in a different setup.

I get message: 
Starting Grabber...
[2016-02-16 20:11:45.754993] [0x000007f17dbaad74] [error]   **XN node not present. One or more of the following nodes could not be enumerated:**

[2016-02-16 20:11:45.755059] [0x000007f17dbaad74] [error]   Failed to initialize openni_acquisition plugin. /home/padeler/work/mbv/release_libs/src/mbv/Acquisition/OpenNIGrabber/OpenNIGrabber.cpp(166): Throw in function void MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: boost::exception_detail::clone_impl<MBV::Core::Exception>
[MBV::Core::Message*] = OpenNI error. Node not present

Entering main Loop.
[2016-02-16 20:11:45.755114] [0x000007f17dbaad74] [error]   Xn function Failed: Input pointer is null!

**Is carmine 1.09 supported. What other sensors can I use?**

thanks for your answer in advance

padeler, 2016-02-16:
As far as I know, carmine is not working with openni 1.x and our sample is configured to work with an openni1.x grabber. You can try the Asus Xtion sensor or the Kinect1. 

Alternately you can grab the images using your own s/w and supply them to the tracker. Just modify the python script to use your own grabber.

didu2225, 2016-03-03:
Hello!
 I'm getting the same error, but I'm trying to make the program work with a Kinect for Xbox One. Should there be any problems with this sensor?

padeler, 2016-03-03:
The Acquisition module supports input from Kinect2 only through the Kinect MS SDK (which is windows only). If you are on windows and have your kinect2 working with MS SDK for Kinect2 you should have no problems running the tracker.

closed 2017-05-05 by padeler