FORTH-ModelBasedTracker/HandTracker #23: RuntimeError: 'output': reserved word

ufoym, 2016-02-24:
I followed the given installation steps, but still got the following error when I ran runme.bat. Where / How can I fix this error?

"Running the Single Hand Tracker script..."
Creating Renderer...
[2016-02-24 15:07:42.696000] [0x00000450] [warning] Read options from environment : unrecognised option 'libs'
[2016-02-24 15:07:42.696990] [0x00000450] [warning] Read options from environment : unrecognised option 'libs'
Traceback (most recent call last):
  File "SingleHandTracking.py", line 32, in <module>
    renderer = Rendering.RendererOGLCudaExposed.get()
RuntimeError: E:\mbv\Rendering\RendererOGL\OpenGLHelpers.cpp(158): Throw in function void __cdecl OGL::ThrowOnCompileLinkError(unsigned int,enum OGL::CompLinkMode,const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = error while compiling file :
ERROR: 0:2: 'output' : reserved word
ERROR: 0:2: 'output' : syntax error syntax error

nkyriazis, 2016-02-24:
Unfortunately, it cannot be fixed in the current release. This is a rare error that we, however, will fix in the next release.

padeler, 2016-02-24:
Can you give us some more information about your setup? (Driver and OpenGL version?) This issue is related to the shader code in the HandTracker renderer and does not appear in most systems.

ufoym, 2016-02-24:
OpenNI SDK: OpenNI-Win64-1.5.7.10-Dev
Sensor Driver: Sensor-Win64-5.1.6.6-Redist
Kinect SDK: KinectSDK-v2.0_1409
CUDA: 7.5
Python: 2.7 x64
OS: Windows 10

padeler, 2016-02-24:
Probably the issue is with your OpenGL and NVidia driver versions.

HelloCentaurus, 2016-03-12:
the same problem,did you fix it?
 File "F:/HandTracker-master/SingleHandTracking.py", line 33, in <module>
    renderer = Rendering.RendererOGLCudaExposed.get()
RuntimeError: E:\mbv\Rendering\RendererOGL\OpenGLHelpers.cpp(158): Throw in function void __cdecl OGL::ThrowOnCompileLinkError(unsigned int,enum OGL::CompLinkMode,const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = error while compiling file :
ERROR: 0:2: 'output' : reserved word 
ERROR: 0:2: 'output' : syntax error syntax error

YuYangguang, 2016-04-08:
I met that issue,too. Do you know the reason of the issue. I guess it is brought by the hardware. But can you give me the hardware requirement? I have changed three devices, all of which reported this error? Can you give me some tips?

HelloCentaurus, 2016-04-08:
Are you Chinese ？I suppose you are，so I will answer your question in Chinese.
具体记得不太清楚了，主要是路径的问题，那个bat文件里设定了环境变量，可能不太行。你可以手动设定环境变量，还有库文件要放在和bat文件放在一个目录下。应该不是硬件问题。

nkyriazis, 2016-04-08:
In your effort @HelloCentaurus to be helpful please be inclusive, too. Not everybody can read Chinese or is able/willing to translate.

@YuYangguang , this is a recurrent error that can and will be fixed on our side. We will provide new binaries with this bug being fixed.

YuYangguang, 2016-04-10:
First of all, I'm sure it is not the issue of the path. I just put the libs and the app  folder on the desktop separately. It also runs well. I suspect it is the issue of hardware. My device is thinkpad, whose Integrated graphics is produced by the intel and discrete graphics is produced by the Nvidia. In the thinkpad device, it occured such error. But when I turned to another device whose graphics are produced by the Nvidia, everything runs well. Maybe the cuda can work well only on the Nvidia GPU??I hope this can help people who are also bothered by this issue.

nkyriazis, 2016-04-11:
The library makes use of CUDA and cannot work on non-CUDA hardware. It also utilizes the primary graphics adaptor.

xiaolei0114, 2017-04-05:
I came across with the same problem. Based on my experiment, it was because the program is utilizing the primary graphics adaptor, and when the primary graphics adaptor is not Nvidia ones, it will cause this problem.
Solved it by setting the preferred graphics processor to the Nvidia one's.

zhuzilin, 2017-09-08:
Thank you! Change the primary graph card works!

closed 2018-02-28 by padeler