FORTH-ModelBasedTracker/HandTracker #24: Question on the structure & definition of PyMBVCore.CameraMeta

HawkRong, 2016-03-01:
Hi,
    I've read the Python script and I found the MBV pretty cool; now I want to try some self-collected data, and I found it necessary to know the details of PyMBVCore.CameraMeta. Would you mind tell me something about this?

Thank You! :-)

padeler, 2016-03-01:
Hello, 

You can initialize a CameraMeta instance by supplying your camera's intrinsics.
For example this is how you could initialize a CameraMeta for the xtion depth sensor:

```python
# xtion calib
cx,cy        = [320.0,240.0]
width,height = [640.0,480.0]
fx,fy        = [570.34,570.34]
        
zNear,zFar = 400.0,10000.0
depthCalib = CameraMeta()       
depthCalib.cameraId=0 # id is usefull on multicamera setups
depthCalib.width  = width
depthCalib.height = height

camera = CameraFrustum()
camera.setIntrinsics(fx,fy,cx,cy,width,height,zNear,zFar)
depthCalib.camera = camera

# use depthCalib as calibration for your camera.
```

HawkRong, 2016-03-02:
It's very helpful. Thank You!

closed 2016-03-10 by nkyriazis