FORTH-ModelBasedTracker/HandTracker #25: import error

WangZixuan, 2016-03-02:
Hello, thanks for your work. I am working with Kinect to do some research.

I received the following import error. 

> MBV_LIBS Environment variable should point to the MBV libraries folder.
> MBV_LIBS D:\MBV_PythonAPI_Win_1.1
> Setting PATH and PYTHONPATH
> "Running the Single Hand Tracker script..."
> Traceback (most recent call last):
>   File "src\SingleHandTracking.py", line 10, in <module>
>     import PyMBVAcquisition as Acquisition
> ImportError: DLL load failed: The specified module could not be found.

If I commented on line 10, I would get this:

>  File "src\SingleHandTracking.py", line 17, in <module>
>     import PyHandTracker as HT

I only had problems with these two DLLs. That is to say, if I had commented these two lines, the 'import' part worked. What happened?
Thank you for your kind attention.

nkyriazis, 2016-03-02:
Have you installed Kinect SDK 2?

WangZixuan, 2016-03-03:
Yes, from https://www.microsoft.com/en-us/download/details.aspx?id=44561

nkyriazis, 2016-03-03:
I trust that you've followed the install instructions.

I suggest using the Dependency Walker (http://www.dependencywalker.com/) to figure out which dlls are not found and lead to the import problem you're having.

WangZixuan, 2016-03-03:
Thank you for your time. I fixed the dependency problems.
But I had the same problem as issue 23 now, which was 

> RuntimeError: 'output': reserved word

As you explained on that issue, It may be with  OpenGL and NVidia driver versions, so are there some recommendations on these driver versions?
Thank you.

nkyriazis, 2016-03-10:
If you find that the error was due to our instructions please let us know so that we fix it.

The driver I'm using on Windows 8 is 353.90.

Please know that the "output" error will be fixed in a next release.

closed 2016-03-10 by nkyriazis

kheito, 2017-06-21:
Can you share on how you solve the line 10 problem? I'm having that problem right now. Thanks in advance

padeler, 2017-06-21:
This is a dependency issue (missing camera drivers/libs). Make sure all the dependencies are properly installed.

kheito, 2017-06-21:
Thanks for the quick response, very much appreciated. We are using KinectV2 as our camera. Btw, do I still need to install OpenNI even though I am using KinectV2 as our device??

[image]

padeler, 2017-06-21:
Yes, the lib was compiled with support for both openni and kinect so you need to have both installed.

kheito, 2017-06-22:
It's working now. Thanks!

GosseMol, 2017-10-11:
I had to add C:\Program Files\OpenNI\Bin64 to my PATH to fix this