FORTH-ModelBasedTracker/HandTracker #27: DLL load failed

HelloCentaurus, 2016-03-12:
The analyzing result of Dependency walker shows these following DLL cannot be opened, and these Dlls are similar.So how can I fix it? Apologize for my poor English. I'm chinese.
API-MS-WIN-CORE-APIQUERY-L1-1-0.DLL
API-MS-WIN-CORE-APPCOMPAT-L1-1-1.DLL
API-MS-WIN-CORE-APPINIT-L1-1-0.DLL
API-MS-WIN-CORE-ATOMS-L1-1-0.DLL
API-MS-WIN-CORE-COMM-L1-1-0.DLL
API-MS-WIN-CORE-CONSOLE-L1-1-0.DLL
API-MS-WIN-CORE-CONSOLE-L2-1-0.DLL
API-MS-WIN-CORE-CRT-L1-1-0.DLL
API-MS-WIN-CORE-DATETIME-L1-1-1.DLL
API-MS-WIN-CORE-DATETIME-L1-1-2.DLL
API-MS-WIN-CORE-DEBUG-L1-1-1.DLL
API-MS-WIN-CORE-DELAYLOAD-L1-1-1.DLL
API-MS-WIN-CORE-ENCLAVE-L1-1-0.DLL
API-MS-WIN-CORE-ERRORHANDLING-L1-1-1.DLL
API-MS-WIN-CORE-ERRORHANDLING-L1-1-3.DLL
API-MS-WIN-CORE-FIBERS-L1-1-1.DLL
API-MS-WIN-CORE-FILE-L1-2-1.DLL
API-MS-WIN-CORE-FILE-L1-2-2.DLL
API-MS-WIN-CORE-FILE-L2-1-1.DLL
API-MS-WIN-CORE-FILE-L2-1-2.DLL
API-MS-WIN-CORE-HANDLE-L1-1-0.DLL
API-MS-WIN-CORE-HEAP-L1-2-0.DLL
API-MS-WIN-CORE-HEAP-L2-1-0.DLL
API-MS-WIN-CORE-HEAP-OBSOLETE-L1-1-0.DLL
API-MS-WIN-CORE-INTERLOCKED-L1-2-0.DLL
API-MS-WIN-CORE-IO-L1-1-1.DLL
API-MS-WIN-CORE-JOB-L1-1-0.DLL
API-MS-WIN-CORE-JOB-L2-1-0.DLL
API-MS-WIN-CORE-KERNEL32-LEGACY-L1-1-1.DLL
API-MS-WIN-CORE-KERNEL32-LEGACY-L1-1-4.DLL
API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-1.DLL
API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-2.DLL
API-MS-WIN-CORE-LARGEINTEGER-L1-1-0.DLL
API-MS-WIN-CORE-LIBRARYLOADER-L1-2-0.DLL
API-MS-WIN-CORE-LIBRARYLOADER-L1-2-1.DLL
API-MS-WIN-CORE-LIBRARYLOADER-L2-1-0.DLL
API-MS-WIN-CORE-LOCALIZATION-L1-2-1.DLL
API-MS-WIN-CORE-LOCALIZATION-L1-2-2.DLL
API-MS-WIN-CORE-LOCALIZATION-L2-1-0.DLL
API-MS-WIN-CORE-LOCALIZATION-OBSOLETE-L1-3-0.DLL
API-MS-WIN-CORE-LOCALIZATION-PRIVATE-L1-1-0.DLL
API-MS-WIN-CORE-MEMORY-L1-1-2.DLL
API-MS-WIN-CORE-MISC-L1-1-0.DLL
API-MS-WIN-CORE-NAMEDPIPE-L1-2-0.DLL
API-MS-WIN-CORE-NAMEDPIPE-L1-2-2.DLL
API-MS-WIN-CORE-NAMESPACE-L1-1-0.DLL
API-MS-WIN-CORE-NORMALIZATION-L1-1-0.DLL
API-MS-WIN-CORE-PATH-L1-1-0.DLL
API-MS-WIN-CORE-PERFCOUNTERS-L1-1-0.DLL
API-MS-WIN-CORE-PRIVATEPROFILE-L1-1-1.DLL
API-MS-WIN-CORE-PROCESSENVIRONMENT-L1-2-0.DLL
API-MS-WIN-CORE-PROCESSSNAPSHOT-L1-1-0.DLL
API-MS-WIN-CORE-PROCESSTHREADS-L1-1-2.DLL
API-MS-WIN-CORE-PROCESSTHREADS-L1-1-3.DLL
API-MS-WIN-CORE-PROCESSTOPOLOGY-L1-2-0.DLL
API-MS-WIN-CORE-PROFILE-L1-1-0.DLL
API-MS-WIN-CORE-PSAPI-ANSI-L1-1-0.DLL
API-MS-WIN-CORE-PSAPI-L1-1-0.DLL
API-MS-WIN-CORE-QUIRKS-L1-1-0.DLL
API-MS-WIN-CORE-REALTIME-L1-1-0.DLL
API-MS-WIN-CORE-REGISTRY-L1-1-0.DLL
API-MS-WIN-CORE-REGISTRY-L1-1-1.DLL
API-MS-WIN-CORE-REGISTRYUSERSPECIFIC-L1-1-0.DLL
API-MS-WIN-CORE-RTLSUPPORT-L1-2-0.DLL
API-MS-WIN-CORE-SHLWAPI-LEGACY-L1-1-0.DLL
API-MS-WIN-CORE-SHLWAPI-OBSOLETE-L1-2-0.DLL
API-MS-WIN-CORE-SIDEBYSIDE-L1-1-0.DLL
API-MS-WIN-CORE-STRING-L1-1-0.DLL
API-MS-WIN-CORE-STRING-L2-1-0.DLL
[... 286 lines cut ...]
EXT-MS-WIN-RTCORE-NTUSER-WINDOW-EXT-L1-1-0.DLL
EXT-MS-WIN-SECUR32-TRANSLATENAME-L1-1-0.DLL
EXT-MS-WIN-SECURITY-CAPAUTHZ-L1-1-0.DLL
EXT-MS-WIN-SECURITY-CHAMBERS-L1-1-0.DLL
EXT-MS-WIN-SECURITY-CREDUI-L1-1-0.DLL
EXT-MS-WIN-SECURITY-EFS-L1-1-0.DLL
EXT-MS-WIN-SECURITY-EFSWRT-L1-1-1.DLL
EXT-MS-WIN-SESSION-USERMGR-L1-1-0.DLL
EXT-MS-WIN-SESSION-USERTOKEN-L1-1-0.DLL
EXT-MS-WIN-SESSION-WINSTA-L1-1-0.DLL
EXT-MS-WIN-SESSION-WTSAPI32-L1-1-0.DLL
EXT-MS-WIN-SETUPAPI-INF-L1-1-0.DLL
EXT-MS-WIN-SHELL-DIRECTORY-L1-1-0.DLL
EXT-MS-WIN-SHELL-PROPSYS-L1-1-0.DLL
EXT-MS-WIN-SHELL-SHELL32-L1-2-0.DLL
EXT-MS-WIN-SHELL-SHLWAPI-L1-1-0.DLL
EXT-MS-WIN-SHELL32-SHELLCOM-L1-1-0.DLL
EXT-MS-WIN-SXS-OLEAUTOMATION-L1-1-0.DLL
EXT-MS-WIN-UI-VIEWMANAGEMENT-L1-1-0.DLL
EXT-MS-WIN-USP10-L1-1-0.DLL
EXT-MS-WIN-WER-UI-L1-1-0.DLL
EXT-MS-WIN-WER-XBOX-L1-1-0.DLL
EXT-MS-WIN-WINRT-STORAGE-L1-1-0.DLL
EXT-MS-WIN-WLAN-ONEXUI-L1-1-0.DLL
IESHIMS.DLL

nkyriazis, 2016-03-13:
I've never seen a list of missing dlls so big. Are you sure this is right?

HelloCentaurus, 2016-03-14:
Thank you ! This problem is fixed.In fact, the list is more longer which displayed in dependency walker.Anyway,now the dlls can be loaded.But another problem is arised.

closed 2016-03-14 by HelloCentaurus

HeliosRa, 2016-04-07:
Please tell me how did you found these dll among so many others As I am also experiencing same problem

HelloCentaurus, 2016-04-11:
I found these dlls with dependency walker,but it turns out it doesn't mater.

hillsonchan, 2016-09-25:
@HelloCentaurus  
Actully! i have the same trouble, i try to find these DLLs also, but some of them can't found, can you help me send a copy for these DLLs to me? my mailbox: qhtec@163.com. thank you!

HelloCentaurus, 2016-09-28:
@hillsonchan 
It's doesn't  mater and it's not a trouble.If you can't run the program, there must be some other trouble.

hillsonchan, 2016-10-07:
@HelloCentaurus  Thanks for your reply!