FORTH-ModelBasedTracker/HandTracker #29: It seems that the tracking model is very differnet from my hand,why?

HelloCentaurus, 2016-03-14:
these following warning OK?
[2016-03-14 19:11:15.527911] [0x00000104] [warning] Read options from environment : unrecognised option 'libs'
[2016-03-14 19:11:15.543520] [0x00000104] [warning] Read options from environment : unrecognised option 'libs'
  Warning! Material file 'F:\FORTH\HANDTRACKER1\media\sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file 'F:\FORTH\HANDTRACKER1\media\cylinder.mtl' not found!
Warning! Material '' not defined in material file.

padeler, 2016-03-14:
Yes you can safely ignore them :)

closed 2016-03-14 by padeler

HelloCentaurus, 2016-03-14:
But the hand model  didn't track my hand?Could you give me some possible reasons

padeler, 2016-03-14:
You must place your hand on the initialization position (shown on the opencv window). 

Once you have you hand on the correct position, press 's' to start tracking. 

You can test it using the included prerecorded oni file.