FORTH-ModelBasedTracker/HandTracker #31: the pause during the process

YuYangguang, 2016-04-03:
I followed the guide to start the programme whose input was the image from the kinnect2. And the computer could follow my gesture. But the problem is that during this process, the  video would pause for a few seconds after working for a few seconds. Woking, pausing, working, and pausing , ...，whihc was disturbing. Is this phenomenon right? Is there anyone who are confronted with this problem? Or just   because of the setup of my conputer?

padeler, 2016-04-04:
It looks like an issue with camera capture. Do you have the same issue when the application is not on tracking mode? 

Can you test the camera with a different grabber/viewer?
Did you have similar issues with the openni grabber using the prerecorded oni file?

YuYangguang, 2016-04-04:
Yes. I have the same issue when the application is not on tracking mode.

I tested my Kinnect camera on Ubuntu 14.04 using the ros kinnect. It worked well.

The program also works well using the prerecorded oni fle.

Now, I'm using Windows10. I don't whether it is brought by the operate system.

padeler, 2016-04-04:
It sounds like it is not related to the tracker. We have tested this with win10 and the Kinect2 sdk and works normally as expected. Maybe you could try running the tracker from pre-recorded kinect2 data.

YuYangguang, 2016-04-04:
I have found  the issue.

It is because the usb3.0 doesn't work properly.

It  is said by the kinnect SDK that the supported usb3.0 detected unknown width.

But I don't know how to solve that. Is there anyone solve the problem like this?

YuYangguang, 2016-04-08:
I found the reason of the issue. It is because of the usb data transformation speed. This issue was solved by changing a usb3.0 hub in superspeed mode.

closed 2016-04-08 by nkyriazis