FORTH-ModelBasedTracker/HandTracker #32: Driver Issues

trevorb23, 2016-04-07:
I am using the Kinect V2 on windows 10 and installed all the necessary prime sense and openni drivers but for some reason everytime I try to open up niviewer it says the device cannot be found. I am almost positive it is a driver issue but I am not exactly sure how to get prime sense completely working on windows 10 with the kinect 2. Any help would be much appreciated

YuYangguang, 2016-04-08:
I  met that issue before.  I found the reason was that my PC didn't have the usb3.0 hub or the usb3.0 hub controller. You may need to change the device

nkyriazis, 2016-04-08:
OpenNI (NiViewer) and Kinect2 are two entirely disparate things. Use the NiViewer with prime sense sensors. Use Kinect SDK2 with the Kinect2.

Still, is there something related to the HandTracker that's not working for you?

closed 2016-07-25 by nkyriazis