FORTH-ModelBasedTracker/HandTracker #34: Why any param in the  file of hand_ringht_low_RH.xml is defined 5 times？

HelloCentaurus, 2016-04-10:
<segment class_id="0" tracking_level="0" version="3">
	<name>palm_pos</name>
	<hook_name>ground</hook_name>
	<link_type>TranslationQuaternion</link_type>
	<trans_X>-200</trans_X>
	<trans_X>-2000</trans_X>
	<trans_X>2000</trans_X>
	<trans_X>1</trans_X>
	<trans_X>0</trans_X>
	<trans_Y>0</trans_Y>
	<trans_Y>-2000</trans_Y>
	<trans_Y>2000</trans_Y>
	<trans_Y>1</trans_Y>
	<trans_Y>0</trans_Y>
	<trans_Z>-900</trans_Z>
	<trans_Z>-2000</trans_Z>
	<trans_Z>2000</trans_Z>
	<trans_Z>1</trans_Z>
	<trans_Z>0</trans_Z>
	<qX>0</qX>
	<qX>-1</qX>
	<qX>1</qX>
	<qX>1</qX>
	<qX>0</qX>
	<qY>0</qY>
	<qY>-1</qY>
	<qY>1</qY>
	<qY>1</qY>
	<qY>0</qY>
	<qZ>1</qZ>
	<qZ>-1</qZ>
	<qZ>1</qZ>
	<qZ>1</qZ>
	<qZ>0</qZ>
	<qW>0</qW>
	<qW>-1</qW>
	<qW>1</qW>
	<qW>1</qW>
	<qW>0</qW>
</segment>
As show above, take the trans_X for example,the number of trans_X  is 5.Maybe the first is default, the second is min and the third is max. But what's the rest two?Can I get a simple answer or just a hint?

nkyriazis, 2016-04-11:
You're correct in your guess. The first additional value corresponds to whether the current parameter is optimizable or is regarded as fixed. The other value is internal and makes no difference in HandTracker. Meddling with the .xml is not advised, however. If you have some specific alteration in mind please ask and we can provide the information on how to achieve it.

closed 2016-04-11 by nkyriazis

HelloCentaurus, 2016-04-12:
Because the hand model is not suit to my hand, I try to reconstruction the hand model using Python in Blender. However, I just made very little progress.
When my hand was moved quickly, it would be tracked imprecisely, and PSO just trapped in  local minimum at this time. Based on observation of the actual hand, maybe we could approximate estimate the position and orientation of the actual hand(just "palm_pos"), then the PSO can search the best "palm_pos" in a narrow region. So the obvious wrong hand hypotheses will be excluded.
Just a suggest, the idea maybe wrong or not easy to achieved. I hope my poor English not hinder my expression.

kheito, 2017-06-25:
How can the PSO successfully estimates the 6D global pose of the hand?? It is directly stated on your journal but it is not explained thoroughly. If the PSO is used to minimize the discrepancy of hypothesized and actual hand, how come it arrive to the values of 3D position, orientation and full articulation of the human hand? I am really confuse. I hope you'll find time to answer my question. Thanks in advance.

padeler, 2017-06-25:
The objective defined in the paper is a function that computes the discrepancy (error) between the observed and a proposed hand pose. 

PSO will test and evolve a number of hypothesized hand poses (particles) for a number of generations and return with the best hand pose. The "best" pose is defined as the one that had the minimum discrepancy with the observation during the optimization process.

kheito, 2017-06-26:
Is that mean that there are already a set of hand pose with a specific output values and it is compared to the input hand pose using PSO?? I'm sorry if my questions bothers you, but I'm really trying my best to understand it. Thanks by the way for the replies, very much appreciated!

padeler, 2017-06-26:
No. PSO generated the hand poses (hypotheses) and uses the objective function to compute their similarity score to the observations.