FORTH-ModelBasedTracker/HandTracker #36: Using the demo with live Kinect

IasonK, 2016-04-18:
Hi,

great work! I am trying to use SingleHandTracking.py to test it directly with data coming live from a Kinect 2. The demo runs fine with the supplied loop.oni file. According to the comments I changed the path to be empty:

```
oniPath = ''
    acq = Acquisition.OpenNIGrabber(True, True, 'media/openni.xml', oniPath, True)
```

The application starts but crashes shortly thereafter with the following error:
...
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
Starting Grabber...
[2016-04-18 15:36:35.542558] [0x00000182] [error]   XN node not present. One or more of the following nodes could not be enumerated:

[2016-04-18 15:36:35.630817] [0x00000182] [error]   Failed to initialize openni_acquisition plugin. E:\mbv\Acquisition\OpenNIGrabber\OpenNIGrabber.cpp(166): Throw in function void __cdecl MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = OpenNI error. Node not present

Entering main Loop.
[2016-04-18 15:36:35.658765] [0x00000182] [error]   Xn function Failed: Input pointer is null!

Do I need to change anything in the openni.xml ? I have installed the SensorKinect093-Bin-Win64-v5.1.2.1.msi driver .

Kind regards.
Iason

padeler, 2016-04-18:
Hello Iason,

It looks like your camera is not setup properly. Try running the NiViewer that comes with the OpenNI installation. You should have the camera connected on a usb bus alone (with no other devices on the same bus) to be able to use it with VGA resolution @ 30fps.

best,
--
pashalis

IasonK, 2016-04-19:
Hi Pashali,

thanks for your answer, I tried the NiViewer64 and I get the following error output:

One or more of the following nodes could not be enumerated:

Device: PrimeSense/SensorKinect/5.1.2.1: The device is not connected!
Device: PrimeSense/SensorV2/5.1.6.6: The device is not connected!
Device: PrimeSense/SensorKinect/5.1.2.1: The device is not connected!
Device: PrimeSense/SensorV2/5.1.6.6: The device is not connected!

I have essentially only the Kinect plugged in, in a USB 3 port, the only other things are mouse and keyboard but they are on USB 2 ports, which I imagine is a seperate bus. I tried a few different USB ports but it doesn't make any difference. The Kinect studio as well as other applications from the Kinect SDK all run fine. 
How do I debug this?

Kind regards.
Iason

nkyriazis, 2016-04-19:
Please have a look at #8 .

IasonK, 2016-04-19:
#8 runs ok, but the hand is not being tracked well. Virtual hand does not follow the motion of the real hand, problems appear when fingers close, rotations, and in general when I move my hand after a few seconds it loses the tracking. 
Also the hand is not red, turning to green like the examples on the website. When the software starts it is already multiple colors. 
Is something going wrong? Some parameters I can adjust?

nkyriazis, 2016-04-22:
The green->red transition was a feature in an older set of binaries. This no longer works this way. In the demo one places a hand in front of the rendering and presses 'S' to start/stop tracking.

To improve tracking, in the expense of processing throughput, try increasing the budget (particles and/or generations). If the processing rate becomes too low for the desired quality you could switch to offline tracking.

closed 2016-04-26 by nkyriazis

KiraRongJin, 2017-07-26:
Virtual hand does not follow the motion of the real hand, problems appear when fingers close, rotations, and in general when I move my hand after a few seconds it loses the tracking.
When the software starts it is already multiple colors. Works bad~

GosseMol, 2017-10-17:
I have the same issue with NiViewer64:

One or more of the following nodes could not be enumerated:

Device: PrimeSense/SensorKinect/5.1.2.1: The device is not connected!
Device: PrimeSense/SensorV2/5.1.6.6: The device is not connected!
Device: PrimeSense/SensorKinect/5.1.2.1: The device is not connected!
Device: PrimeSense/SensorV2/5.1.6.6: The device is not connected!

How did you solve this? Issue #8 seems unrelated to this issue. Did the link change or am I missing something?