FORTH-ModelBasedTracker/HandTracker #38: Tracking the Meshes of the Hand

Freshy123, 2016-04-21:
Good Afternoon,

I am with a group at Rochester Institute of Technology and we have been working with your "SingleHandTracking.py". We have been looking for a way to track the individual meshes located on the fingertips of the hand rendering. We are currently able to print out a converted numpy array of the variable currentHandPose and from this we obtain 27 values. We are wondering what these values represent, and if they can help us in tracking the position of the individual fingertips on the hand.

We were also wondering how you were able to create object files and render them with the hand in the videos you provided. Could we write something in your python code to integrate .obj files within your code. The reason we ask this is that we aim to try to track the hand as it holds an object, and occlusion one of our concerns. We are wondering how we could incorporate our own .obj files, since the object the hand holds is irregular in nature. Would you be able to recommend any other methods of dealing with occlusion?

Very Respectfully,

Emmanuel Palad, Kaitlyn Emerson, Dominic Curcione
Rochester Institute of Technology
716-421-4181
dpc9190@rit.edu

nkyriazis, 2016-04-22:
It is in our plans to release code showing hand-object 3D interaction tracking. Please stay tuned.

trevorb23, 2016-04-26:
Hello,

I am also a part of the research group at Rochester Institute of Technology utilizing this code. Our project that may incorporate this code is to be finished within the next few weeks so we are a bit strapped for time. Is there any way we could get into more direct contact with you?

Thank you for your time

Freshy123, 2016-04-26:
Trevor,

Make sure you coordinate with EJ and Katie they have a good idea as to what
we need from these individuals. Also the git hub might be your best chance
to contact these individuals.

-Dom

nkyriazis, 2016-04-27:
With respect to what the 27 values mean I suggest you look at #16 . As for the hand-object tracking, our plans go beyond your time limitations. The present library may be used for specific scenarios of hand-object tracking. A library that targets the problem more appropriately, however, will be released later in time. Still, if you think you can specify what is you're trying to achieve, it might be possible that we could help you do this with the present library.

trevorb23, 2016-04-28:
We are trying to utilize your program to track critical points on a hand, specifically fingertips, as one holds objects with irregular geometry, roughly the size of a baseball known as glavins. Occlusion is our biggest issue but it seems that your program has the capabilities to overcome this. It is important to note that we have .obj files containing the geometries of these objects. Any help you could provide towards this goal would be greatly appreciated.

padeler, 2016-05-05:
Hello,

I committed a new script in src called *HandObjectTracking.py*. It is using the BMVC11 objective to track two objects. You will see that the changes compared to the original single hand tracking script are minimal. 

Since this is based on the *BMVC11 single hand tracking objective*, it does not account for collisions and does not know about the "two" objects. It is just optimizing for 34 dimmensions. Similarly you could extend this to two hand tracking.

I also created an example dataset called hand-object.oni. You can use it for testing. The decoder is reading a single obj file which you can change to your model. 

let me know if you have any questions or issues running the script.

closed 2016-07-25 by nkyriazis