FORTH-ModelBasedTracker/HandTracker #41: No module names PyMBVCore

iamlegolas, 2016-06-10:
I'm using the Kinect VERSION 1 on Windows 7. I followed all the steps. In the end when I execute the batch file, I get the error that's attached. Any assistance is welcomed. 

Thankyou
[image]

padeler, 2016-06-13:
Hello,

The MBV libs will not work on windows7, since they are linked against the Kinect2 libraries which are only available on windows8+.

Armixy, 2016-06-22:
[image]

The same error, ran on ubuntu 16.04

Armixy, 2016-06-22:
where is libCore.so?

padeler, 2016-06-22:
FORTH/HANDTRACKER/libs/libCore.so

Armixy, 2016-06-22:
[image]

padeler, 2016-06-22:
As i said in my previous reply, set the LD_LIBRARY_PATH first (before running ldd), to check the current folder i.e.: export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:.

This will fix the "not found" for libraries that are packaged with the tracker like boost and openmesh.

Also please send the whole output of ldd.

Armixy, 2016-06-22:
OK, I will try it, thx a lot.

closed 2016-07-25 by padeler

ashokbugude, 2016-07-29:
Same issue here . Below is my terminal screen output

zayyon@zayyon-MS-7693:~/zayyon/HandTracker-master$ ./runme.shMBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS
Setting LD_LIBRARY_PATH and PYTHONPATH
Running the Single Hand Tracker script...
Traceback (most recent call last):
  File "src/SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: No module named PyMBVCore

padeler, 2016-07-29:
You need to set the MBV_LIBS variable to point to the mbv libraries folder as the script explains.

ashokbugude, 2016-07-29:
I have set it, but getting  errors as shown below

zayyon@zayyon-MS-7693:~/zayyon/HandTracker-master$ export MBV_LIBS=/home/zayyon/zayyon/MBV_PythonAPI_Linux_1.1
zayyon@zayyon-MS-7693:~/zayyon/HandTracker-master$ ./runme.shMBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS /home/zayyon/zayyon/MBV_PythonAPI_Linux_1.1
Setting LD_LIBRARY_PATH and PYTHONPATH
Running the Single Hand Tracker script...
Traceback (most recent call last):
  File "src/SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: /home/zayyon/zayyon/MBV_PythonAPI_Linux_1.1/libs/libCore.so: undefined symbol: _ZN2cv9ExceptionC1EiRKSsS2_S2_i

ashokbugude, 2016-07-29:
Below is the complete output of ldd libCore.so

	linux-vdso.so.1 =>  (0x00007ffc2c9d0000)
	libOpenMeshCore.so.3.1 => not found
	libcudart.so.7.5 => /usr/local/cuda/lib64/libcudart.so.7.5 (0x00007f4b643ce000)
	libboost_system.so.1.57.0 => not found
	libboost_filesystem.so.1.57.0 => not found
	libboost_thread.so.1.57.0 => not found
	libboost_serialization.so.1.57.0 => not found
	libboost_program_options.so.1.57.0 => not found
	libboost_log.so.1.57.0 => not found
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4b641ae000)
	libtbb.so.2 => /usr/lib/x86_64-linux-gnu/libtbb.so.2 (0x00007f4b63f6e000)
	libopencv_core.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 (0x00007f4b63b3e000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4b637b6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4b634a6000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4b6328e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4b62ebe000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4b62cb6000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f4b62aae000)
	/lib64/ld-linux-x86-64.so.2 (0x000055a4c858a000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f4b6288e000)
	libGL.so.1 => /usr/lib/nvidia-361/libGL.so.1 (0x00007f4b625fe000)
	libGLX.so.0 => /usr/lib/nvidia-361/libGLX.so.0 (0x00007f4b623c6000)
	libGLdispatch.so.0 => /usr/lib/nvidia-361/libGLdispatch.so.0 (0x00007f4b620de000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f4b61d9e000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f4b61b86000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f4b6195e000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4b61756000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4b6154e000)

Taboen, 2017-10-17:
I have this same issue on windows 10, with a fresh windows 10 install. I'm guessing I need to install something, but I don't know what it should be. I followed the steps mentioned.

> D:\Users\user\InsignityAR\FORTH\app>runme.bat
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS
Setting PATH and PYTHONPATH
"Running the Single Hand Tracker script..."
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: No module named PyMBVCore
(C:\Users\user\Anaconda2) D:\Users\user\InsignityAR\FORTH\app>

Depth3D, 2017-12-06:
the library is not frrendly....