FORTH-ModelBasedTracker/HandTracker #44: Python to Matlab

ma581, 2016-07-08:
Hi FORTH :)

I am now trying to feed the coordinates of the model hand calculated during hand tracking into MATLAB (Windows 10). I am using Matlab to execute operating system command using the function `<system()>`. 

The error I get seems to be related to the hand tracking libraries. Kindly could you help me out? Thanks!

Error message in MATLAB
```
Setting PATH and PYTHONPATH 
"Running the Single Hand Tracker script..." 
Creating Renderer... 
Traceback (most recent call last): 
  File "C:\Users\store\Code\07_CBAS\HandTracker\src\SingleHandTracking.py", line 60, in <module> 
    ht = HT.HandTrackerLib(2048, 2048, 64, 64, erenderer) 
RuntimeError: E:\mbv\Decoding\GenericDecoder\GenericDecoder.cpp(1320): Throw in function void __cdecl MBV::Decoding::GenericDecoder::loadFromFile(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,bool) 
Dynamic exception type: struct MBV::Core::Exception 
[struct MBV::Core::Message * __ptr64] = could not open file for reading 
[struct MBV::Core::OffendingCode * __ptr64] = ifs.good()
```

padeler, 2016-07-11:
It looks like the working directory for your script is not setup properly. It should be the same as the directory where the ".bat" file is.

ma581, 2016-07-13:
Thanks that did the trick! :D

ehewage, 2016-07-13:
My goal with combining the hand tracker is to be able to control a robotic hand that is controlled through Matlab. 

I am able to run SingleHandTracking.py from Matlab but I am not sure about the best way to enable Matlab to access the hand tracking variables that are updated in SingleHandTracking.py in real-time. 

At present, in Matlab, I am using:  
```[status, result] = system('C:\Users\store\Code\07_CBAS\HandTracker\runme2.bat')```
And in SingleHandTracking.pyI have added a line: 
```print("Matrix = ", matrix)``` which has decoded model hand coordinates. 
However, Matlab is not able to access the variable (called *matrix* ) live (but only after I have stopped the hand tracker).  

Is there a suitable method to enable live update of the variable within Matlab? 
Is there a method to pipe the Python variable to Matlab in Windows? 

Thanks!!

ma581, 2016-07-13:
My goal with combining the hand tracker is to be able to control a robotic hand that is controlled through Matlab. 

I am able to run SingleHandTracking.py from Matlab but I am not sure about the best way to enable Matlab to access the hand tracking variables that are updated in SingleHandTracking.py in real-time. 

At present, in Matlab, I am using:  
```[status, result] = system('C:\Users\store\Code\07_CBAS\HandTracker\runme2.bat')```
And in SingleHandTracking.pyI have added a line: 
```print("Matrix = ", matrix)``` which has decoded model hand coordinates. 
However, Matlab is not able to access the variable (called *matrix* ) live (but only after I have stopped the hand tracker).  

Is there a suitable method to enable live update of the variable within Matlab? 
Is there a method to pipe the Python variable to Matlab in Windows? 

Thanks!!

nkyriazis, 2016-07-25:
InterProcess Communication seems indeed the suitable route. You can try sending the data in binary form (27 doubles) and reading it back in matlab in binary form too, so no parsing is needed.

closed 2016-07-25 by nkyriazis