FORTH-ModelBasedTracker/HandTracker #45: OpenNI error. Function Failed

liming0791, 2016-07-24:
Hello, when I run `./runme.sh` I get this error message:

```
Starting Grabber...
[2016-07-24 22:30:00.887199] [0x000007f0828f7d74] [error]   Xn function Failed: The file is corrupted!
[2016-07-24 22:30:00.887262] [0x000007f0828f7d74] [error]   Failed to initialize openni_acquisition plugin. /home/padeler/work/mbv/release_libs/src/mbv/Acquisition/OpenNIGrabber/OpenNIGrabber.cpp(171): Throw in function void MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: boost::exception_detail::clone_impl<MBV::Core::Exception>
[MBV::Core::Message*] = OpenNI error. Function Failed

```

Below is all the output,  hope it is useful:

```
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS /home/liming/MBV_PythonAPI_Linux_1.1/
Setting LD_LIBRARY_PATH and PYTHONPATH
Running the Single Hand Tracker script...
Creating Renderer...
[2016-07-24 22:30:00.396921] [0x000007f0828f7d74] [warning] Read options from environment : unrecognised option 'libs'
[2016-07-24 22:30:00.397027] [0x000007f0828f7d74] [warning] Read options from environment : unrecognised option 'libs'
  Warning! Material file '/home/liming/HandTracker-master/media/sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file '/home/liming/HandTracker-master/media/cylinder.mtl' not found!
Warning! Material '' not defined in material file.
Variances:  [10.0, 10.0, 10.0, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]
Low Bounds:  [-2000.0, -2000.0, -2000.0, -1.0, -1.0, -1.0, -1.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 2.5, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0]
High Bounds:  [2000.0, 2000.0, 2000.0, 1.0, 1.0, 1.0, 1.0, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.0, 3.140000104904175, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874]
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
Starting Grabber...
[2016-07-24 22:30:00.887199] [0x000007f0828f7d74] [error]   Xn function Failed: The file is corrupted!
[2016-07-24 22:30:00.887262] [0x000007f0828f7d74] [error]   Failed to initialize openni_acquisition plugin. /home/padeler/work/mbv/release_libs/src/mbv/Acquisition/OpenNIGrabber/OpenNIGrabber.cpp(171): Throw in function void MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: boost::exception_detail::clone_impl<MBV::Core::Exception>
[MBV::Core::Message*] = OpenNI error. Function Failed

Entering main Loop.
Traceback (most recent call last):
  File "src/SingleHandTracking.py", line 127, in <module>
    c = clbs[0]
IndexError: Index out of range

```

Besides, I use `Ubuntu 14.04` , `OpenNI` and a `kinect v1`, when I run `NiViewer`, it works fine.

padeler, 2016-07-25:
Hello,

Typically the issue here is a missconfigured openni installation. However since the NiViewer is working fine there are two possible couses:

1. You have 32bit openni installation (64bit is required)

2. Your camera is connected on a USB bus that is used by other devices as well. The NiViewer starts the camera on QVGA resolution for both streams while the HandTracker starts is on VGA resolution. If the USB bus is used by another device the bandwidth is not enough for the streaming for 2 VGA streams at 30fps. 

Try setting the NiViewer to VGA for both streams and see if the camera still works normally. 
Try changing the USB port where the camera is connected. Make sure it is the only device on the bus.

rongfeng-china, 2016-08-20:
Hi, @padeler 

I have the same "Openni Error" and NiViewer is running well...
My system setting is Ubuntu 16.04, [openni 1.5.4](https://github.com/OpenNI/OpenNI), and my camera is Asus Pro.

I only used one USB port and tried "lsusb" command:
[image]

Do you know what problem may cause the error? Thanks...
For the VGA setting for NiViewer, I'm not sure how to do that... Could you explain more if this is the problem... Thank you!

gsiriani, 2016-11-20:
Hi, I'm having the exact same issue here.
I also noticed there is a weird path in the error message:
/home/padeler/work/mbv/release_libs/src/mbv/Acquisition/OpenNIGrabber/OpenNIGrabber.cpp
It's in the second error in liming0791's reported output.
Shouldn't that path point to the local installation? 
Thanks

padeler, 2016-11-24:
@gsiriani can you check if the NiViewer works when both streams are set to VGA resolution (see second bullet on my original reply). The error message is normal. It refers to the source not the local installation.

gsiriani, 2016-12-01:
@padeler I set both streams to VGA. The camera still works fine, but i keep getting the same issue when executing the script

padeler, 2016-12-02:
Do you get the same issue when you try to open the camera and when you try to open an "oni" recording?

yayo13, 2016-12-21:
@padeler I have the same issue with @liming0791 . i've tried on windows & Ubuntu with different usb port but all get OpenNI error. Besides, NiViewer works fine...
And, could u teach me how to set NiViewer to VGA res?

MarekStojecki, 2017-01-19:
@yayo13   open NiViewer and click right mouse button you should see a menu.
@padeler I have the same problem as gsiriani. NiViewer works fine although I get the "OpenNI error. Function Failed" when I try to open an "oni" recording. Could you please provide any suggestion how to fix it? Many Thanks.

padeler, 2017-01-20:
@MarekStojecki Maybe you don't have the correct version of openni installed? I have never encountered an issue while grabbing from an "oni" file. 

If you cannot get your camera to work with our grabber you can always supply the depth and rgb images directly to the tracker, check the python script comments for details.

MarekStojecki, 2017-01-20:
@padeler Ok. Thank you. I'm using Openni 1.5.4 but probably I'll write my own grabber as you suggested.

xiaolei0114, 2017-04-13:
Same issue here.
Solved it by modifying the code in line 90 from
oniPath = 'loop.oni'
to 
oniPath = ''
and it worked.
Maybe give it a try will help.

Best luck

closed 2017-04-21 by padeler