FORTH-ModelBasedTracker/HandTracker #47: Error during installation

ashokbugude, 2016-08-04:
Hi I am installing in Ubuntu Linux 16.04. and I get errors as shown below

zayyon@zayyon-MS-7693:~/zayyon/HandTracker-master$ ./runme.shMBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS /home/zayyon/zayyon/MBV_PythonAPI_Linux_1.1/
Setting LD_LIBRARY_PATH and PYTHONPATH
Running the Single Hand Tracker script...
Traceback (most recent call last):
  File "src/SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: /home/zayyon/zayyon/MBV_PythonAPI_Linux_1.1//libs/libCore.so: undefined symbol: _ZN2cv9ExceptionC1EiRKSsS2_S2_i

padeler, 2016-08-04:
We have not tested with Ubuntu 16.04 yet. It should work if you have the correct versions of the required libs. 

From the error it looks like a version problem with opencv (cv::Exception). 
The packaged mbv libs (including libCore.so) are compiled with opencv 2.4.8.

ashokbugude, 2016-08-04:
Do I need to install opencv 2.4.8 ?

padeler, 2016-08-04:
OpenCV 2.4.8 is the version packaged with ubuntu 14.04.

I believe any 2.4.x version should work.

ashokbugude, 2016-08-04:
My opencv version is 2.4.9.1

padeler, 2016-08-04:
Strange, since cv::Exception should be the same in all 2.4.x version. I any case, you could try using 2.4.8 or test on ubuntu 14.04.

olkido, 2016-08-04:
I was having the same issue on ubuntu 15.10, I think it might be the same issue reported here https://github.com/FORTH-ModelBasedTracker/HandTracker/issues/2, I gave up and went for an earlier Ubuntu version, as  I was not able to get OpenCV with Python to compile with an older gcc version as suggested there.

victor1234, 2017-01-04:
I have the similar problem too with 2.4.8 on Ubuntu 16.04. Could you tell with what library version you complied project?

closed 2017-09-11 by padeler