FORTH-ModelBasedTracker/HandTracker #48: Using Hand Tracker with pre-recorded sequences

olkido, 2016-08-04:
Hi,
I would like to use the Hand Tracker (for now just the SingleHandTracking.py) with some RealSense sequences that I have previously recorded and saved to image files. Ideally I would like to simply load opencv images, i.e. not go through an .oni file as suggested here  https://github.com/FORTH-ModelBasedTracker/HandTracker/issues/42 . Is this possible? Could you perhaps help me out with some tips on how to set the calibration data (variable clbs if I understand correctly) (formats etc) and the image data (variable imgs)? 
Thanks very much.

nkyriazis, 2016-08-08:
Image acquisition is performed at [line 122](https://github.com/FORTH-ModelBasedTracker/HandTracker/blob/master/src/SingleHandTracking.py#L122). You can replace the default acquisition with any code that can produce a pair of images (RGB and depth) and some calibration information.

Consider replacing with the following:
```python
import cv2 as cv

# It is assumed that both RGB and depth streams have the same dimensions as images
# The calibration used corresponds to the depth stream
imgs = [ cv.imread(pathToDepthImage, cv.IMREAD_UNCHANGED), # Make sure 16bit loading is perserved
         cv.imread(pathToRGBImage) ]
cfr = Core.CameraFrustum()
cfr.OpenCV_setIntrinsics(fx, fy, cx, cy) # assuming some focal lengths fx, fy and principal point cx, cy.
cfr.OpenCV_setExtrinsics(translation, rotation) # assuming a 3D translation and rodrigues rotation. Optional.
clb = Core.CameraMeta(cfr, width, height) # assuming the width and height of the images in pixels
```

olkido, 2016-08-09:
Thanks @nkyriazis ! I'm getting the following

  File "src/SingleHandTracking.py", line 143, in <module>
    cfr.OpenCV_setIntrinsics(fx, fy, cx, cy) # assuming some focal lengths fx, fy and principal point cx, cy.
Boost.Python.ArgumentError: Python argument types in
    CameraFrustum.OpenCV_setIntrinsics(CameraFrustum, float, float, float, float)
did not match C++ signature:
    OpenCV_setIntrinsics(MBV::Core::CameraFrustum {lvalue}, cv::Mat, cv::Size_<int>, float, float)

Could that be a wrong OpenCV  version or am I  calling something wrong? (I am very new to python and I'm not sure I am parsing this error right - I simply gave numerical values to fx ,fy, cx, cy)

nkyriazis, 2016-08-09:
My bad. I updated the code to better reflect the API. If the nature of the extra parameters is still not clear I can come back with more details.

closed 2016-08-11 by nkyriazis