FORTH-ModelBasedTracker/HandTracker #51: Windows 8.1: Error importing PyMBVCore

Jsaidias, 2016-11-08:
Hello.

I followed all the instructions about installing the correct drivers and programs to run the Hand Tracker provided in a Kinect v2 and I get this error after opening the `runme.bat` file:

`C:\3D_Hand_Tracking\HandTracker>runme.bat
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS
Setting PATH and PYTHONPATH
"Running the Single Hand Tracker script..."
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: No module named PyMBVCore`

The environment variable `MBV_LIBS` was set into the root of the folder provided by the package downloaded (for Windows) with the libraries.

My OS: Windows 8.1 Pro 64 bits
GPU card: NVIDIA GEForce GT 635M (CUDA compute capability = 2.1) with OpenGL version 4.4
6GB RAM

Programs/Drivers Installed: 
Python 2.7 (64 bits)
Anaconda2 4.2.0 64 bits
OpenNI 1.5.7.10 (64 bits)
PrimeSensor
Visual C++ 64 bits for VStudio 2013
Kinect 2 SDK
CUDA 8.0 Driver

Please, give me some advice.

Jsaidias, 2016-11-15:
Issue solved.

closed 2016-11-15 by Jsaidias

maikelronnau, 2017-01-03:
What was the solution? Please, share...

mecastro, 2017-03-29:
I'm currently having this problem pls help

padeler, 2017-03-29:
If you are having the same issue @mecastro and @maikelronnau, you are not setting the environment variable properly. When the script starts the second line should display the actual path in MBV_LIBS.

mecastro, 2017-03-29:
by that you mean this right "set MBV_LIBS=c:\Users\User\Documents\FORTH\HANDTRACKER"?

padeler, 2017-03-29:
Yes, It must point to the folder were you unziped the binaries package.

Note: The environment variables set in the terminal are not permanent and hold only for that specific terminal. If you want to make the permanent in windows you should go through the system settings menus.

mecastro, 2017-03-29:
@padeler already set the environment variables its opening but for only 0.5s then it close again here the screenshot 

[image]

padeler, 2017-03-29:
since the environment var is not printed next to MBV_LIBS it is not set properly. Open the script and see that it is just printing it. Make sure the var is properly set.

mecastro, 2017-03-29:
ok, here is what I've done 

[image]

just want to know if I'm doing it correctly

mecastro, 2017-03-29:
@padeler just want to ask if i need to change something in the script

[image]

Thanks

mecastro, 2017-03-29:
@padeler I already manage to set the environment variable nowmy problem is numpy.core.multiarray.failed to import

[image]

padeler, 2017-03-29:
Check the instructions here: https://github.com/FORTH-ModelBasedTracker/HandTracker
You need to properly install numpy.

mecastro, 2017-03-30:
@padeler do i need to connect the kinect  for it to be imported?

i already installed it using the anaconda command prompt and the regular command prompt i even updated it but the problem still persist

KiraRongJin, 2017-08-01:
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: DLL load failed: %1 is not a valid Win32 application.

how to solve it?

padeler, 2017-08-01:
This is probably an unmet dependency. Make sure that all required libs are installed (see the project README file for details). If you still have the issue use a dependency walker to see which dll is missing.