FORTH-ModelBasedTracker/HandTracker #52: kinect2input branch runs without tracking

DonnyVerduijn, 2017-01-23:
Hello,

When running the program, the hand is not being tracked.
However, the visualization is affected by the input of the stream.

How should i proceed?

DonnyVerduijn, 2017-01-24:
It seems that the program cannot properly locate the hand in order to generate the model. It is possible to track the hand but it needs interaction from the user to find the hand. This is not usually the case right?

padeler, 2017-01-24:
In order to initialize the tracking you need to place your hand over the super-imposed hand model shown on the "Hand Tracker" window and press 's'. The method requires a good initial position and will keep tracking your hand from that point on (or until the track is lost).

You can try the pre-recorded dataset named loop.oni.

DonnyVerduijn, 2017-01-24:
Thank you. I assumed the program automatically localized the hand before super-imposing the model.

closed 2017-01-24 by DonnyVerduijn