FORTH-ModelBasedTracker/HandTracker #54: Error during installation （window10）

RenFeiTemp, 2017-03-20:
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS E:\handTrack\MBV_PythonAPI_Win_1.1
Setting PATH and PYTHONPATH
"Running the Single Hand Tracker script..."
Creating Renderer...
[2017-03-20 17:50:38.345394] [0x0000018f] [warning] Read options from environment : unrecognised option 'libs'
[2017-03-20 17:50:38.346417] [0x0000018f] [warning] Read options from environment : unrecognised option 'libs'
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 33, in <module>
    renderer = Rendering.RendererOGLCudaExposed.get()
RuntimeError: E:\mbv\Rendering\RendererOGL\OpenGLHelpers.cpp(158): Throw in function void __cdecl OGL::ThrowOnCompileLinkError(unsigned int,enum OGL::CompLinkMode,const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = error while compiling file :
ERROR: 0:2: 'output' : reserved word
ERROR: 0:2: 'output' : syntax error syntax error

padeler, 2017-03-20:
You are probably using installing on a machine with more than one GPUs. Please disable the secondary GPU and keep only the NVidia GPU while running MBV.

closed 2017-03-20 by padeler

RenFeiTemp, 2017-03-20:
thanks,i have solved this problem.
then,i meet another problem：
  Warning! Material file 'E:\handTrack\HandTracker-master\media\sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file 'E:\handTrack\HandTracker-master\media\cylinder.mtl' not found!
Warning! Material '' not defined in material file.
Variances:  [10.0, 10.0, 10.0, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]
Low Bounds:  [-2000.0, -2000.0, -2000.0, -1.0, -1.0, -1.0, -1.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 2.5, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0]
High Bounds:  [2000.0, 2000.0, 2000.0, 1.0, 1.0, 1.0, 1.0, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.0, 3.140000104904175, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874]
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
Starting Grabber...
[2017-03-20 20:11:03.210154] [0x00000272] [error]   Xn function Failed: The file is corrupted!
[2017-03-20 20:11:03.248053] [0x00000272] [error]   Failed to initialize openni_acquisition plugin. E:\mbv\Acquisition\OpenNIGrabber\OpenNIGrabber.cpp(171): Throw in function void __cdecl MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = OpenNI error. Function Failed

Entering main Loop.
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 127, in <module>
    c = clbs[0]
IndexError: Index out of range

RenFeiTemp, 2017-03-20:
and i have  installed  Sensor-Win64-5.1.6-Redist and OpenNI-Win64-1.5.7-Dev , but i can not run NiSimpleViewer64.exe.

padeler, 2017-03-27:
The example loop.oni cannot be opened for some reason (the logs you posted say file corrupted). Try giving another oni file or opening the camera. Check the python source code for details on how to do that. 

Also please post separate questions in separate issues to help other people who might have similar problems find an answer easier.