FORTH-ModelBasedTracker/HandTracker #58: Output Syntax Error

mecastro, 2017-03-30:
I've made a progress and the numpy is now importing perfectly but i encountered another problem

[image]

Pls help

mecastro, 2017-03-30:
this is a much better screenshot 

[image]

padeler, 2017-03-30:
You probably have two graphics cards on your laptop. You must disable the intel one kai keep only the nvidia card enabled. Check issue #23 for more details.

closed 2017-03-30 by padeler

mecastro, 2017-03-30:
@padeler i only have  1 graphics, although its only the onboard graphic cards, is that ok 

[image]

padeler, 2017-03-30:
You need an NVidia GPU to run the tracker.