FORTH-ModelBasedTracker/HandTracker #60: Index out of range

mecastro, 2017-04-01:
ok i have an nvidia now and i encountered a new error here is the screencap 

[image]

hope you can help me thank you

mecastro, 2017-04-01:
NVM already fixed the issue 

one last issue tho how to speed up the tracking

padeler, 2017-04-03:
The tracker's performance is bounded by your GPU and to a (much) lesser degree by your CPU. You should be able to get 25+ fps using a NVidia GTX 580 or better on an modern i3 CPU.

closed 2017-04-03 by padeler

asd2846531, 2017-04-05:
Hi,
I have the same Index out of range issue.
Can you provide your solution to the issue?

Many thanks

padeler, 2017-04-05:
Hello, the actual error is higher up: The OpenNI cannot read the recording file. Try using a different oni recording or just open directly the camera. Check the comments in the python file for more information.