FORTH-ModelBasedTracker/HandTracker #63: Help!AttributeError: 'module' object has no attribute 'Kinect2MSGrabber'

myyAlan, 2017-05-14:
When i running the runme.sh,i get a problem about the attribute of module,but i dont know what is kind of error it is and what should i do to avoid the error?Thank you very much!

`  Warning! Material file '/home/alan/normalfiles/kinect/fingertip/HandTracker-Kinect2Input/media/sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file '/home/alan/normalfiles/kinect/fingertip/HandTracker-Kinect2Input/media/cylinder.mtl' not found!
Warning! Material '' not defined in material file.
Variances:  [10.0, 10.0, 10.0, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]
Low Bounds:  [-2000.0, -2000.0, -2000.0, -1.0, -1.0, -1.0, -1.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 2.5, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0]
High Bounds:  [2000.0, 2000.0, 2000.0, 1.0, 1.0, 1.0, 1.0, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.0, 3.140000104904175, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874]
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
Starting Grabber...
Traceback (most recent call last):
  File "src/SingleHandTracking.py", line 87, in <module>
    Ch = Acquisition.Kinect2MSGrabber.Channels
AttributeError: 'module' object has no attribute 'Kinect2MSGrabber'
`

padeler, 2017-05-15:
Hello, 

The grabber for the Kinect v2 works only on windows, using the MS-Kinectv2 SDK drivers. In linux you can either use one of the supported cameras (Asus Xtion, Kinect v1), or provide your own input with your own grabber. For the later option check the python code comments for details.

closed 2017-05-15 by padeler

myyAlan, 2017-05-15:
I had read the code comments  but i dont know what should  i do .  Can you tell how can I get a grabber for the Kinect V2 or maybe you can offer me the code?Thang you very much!

padeler, 2017-05-15:
We do not have a Linux grabber for Kinect2. 

You can adapt code from the libfreenect2 (https://github.com/OpenKinect/libfreenect2) but we have not tested it on linux. 

That said, the tracker is camera agnostic so if you can provide a registered RGB-D pair of images along with their intrinsic calibration parameters it will track your input.

myyAlan, 2017-05-16:
Well,Thank you,I am a college student and I have my own code for hand tracking for kinect2  ,but it cant work well when my hands move fastly. Can you offer me your code for a reference?Or maybe you can tell 
me how to use the tracker now that I have the images and what is kind of calibration parameters the tracker needed?Thank you again for your precious time!

padeler, 2017-05-16:
You can check the discussion on #48 for code examples on how to provide raw images and calibration parameters to the tracker.