FORTH-ModelBasedTracker/HandTracker #65: [Emergency]  I cannot understand the handcopied errors.....

KiraRongJin, 2017-07-22:
C:\Users\mmlab\Documents\FORTH\HANDTRACKER>runme.bat
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS C:\Users\mmlab\Documents\FORTH\HANDTRACKER
Setting PATH and PYTHONPATH
"Running the Single Hand Tracker script..."
Creating Renderer...
[2017-07-21 20:16:05.842607] [0x000000fa] [warning] Read options from environment : unrecognised option 'libs'
[2017-07-21 20:16:05.842607] [0x000000fa] [warning] Read options from environment : unrecognised option 'libs'
  Warning! Material file 'C:\Users\mmlab\Documents\FORTH\HANDTRACKER\media\sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file 'C:\Users\mmlab\Documents\FORTH\HANDTRACKER\media\cylinder.mtl' not found!
Warning! Material '' not defined in material file.
Variances:  [10.0, 10.0, 10.0, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]
Low Bounds:  [-2000.0, -2000.0, -2000.0, -1.0, -1.0, -1.0, -1.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 2.5, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0]
High Bounds:  [2000.0, 2000.0, 2000.0, 1.0, 1.0, 1.0, 1.0, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.0, 3.140000104904175, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874]
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
Starting Grabber...
[2017-07-21 20:17:10.769639] [0x000000fa] [error]   Xn function Failed: The file is corrupted!
[2017-07-21 20:17:10.770643] [0x000000fa] [error]   Failed to initialize openni_acquisition plugin. E:\mbv\Acquisition\OpenNIGrabber\OpenNIGrabber.cpp(171): Throw in function void __cdecl MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = OpenNI error. Function Failed

Entering main Loop.
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 127, in <module>
    c = clbs[0]
IndexError: Index out of range

padeler, 2017-07-24:
The OpenNI cannot open the oni recording that you provide in the script. You can try to open the camera directly (by setting an empty string to the oniFile variable) or set it to another oni recording. Also please make sure you camera is setup correctly (i.e the NiViewer can open the camera and grab frames). 

For more details on using the handtracker script check the comments inside the .py file.

closed 2017-07-24 by padeler

KiraRongJin, 2017-07-25:
It is not just because of what you suggested. It is because of camera version, since I am using Kinect sdk 2. So I used another .py file instead.