FORTH-ModelBasedTracker/HandTracker #66: Module PyMBVCore not found (Win10)

iDaN5x, 2017-07-26:
Hello,

I'm trying to run the HandTracker demo application provided here, with no success.
I have installed all the required dependencies, and set the MBV_LIBS variables as required.
However, when running the `runme.bat` file, python throws out an ImportError for PyMBVCore...

[image]

Can you help me solve this problem?

Thanks.

iDaN5x, 2017-07-29:
@padeler

_Sent from my Samsung SM-G955F using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

padeler, 2017-07-31:
Hello,

PyMBVCore is located inside the python_libs folder. From your screenshot it looks like you have set the environment MBV_LIBS variable to point to the correct folder. 

Maybe you have a missing dependency?

KiraRongJin, 2017-07-31:
I have the same problem on Win8 and the following versions:
numpy1.13.1
scipy0.16.1
matplotlib2.0.2
opencv2.4.13.2

lyyiangang, 2017-08-01:
I meet the same problem. It says "DLL load failed"  but I can't get more details about this failure. I don't know what exact DLL is missed. 
Note: I am using the graphical card of AMD instead of NVIDIA.  Is it the root cause of this import error??
Could you please give any tips?
Thanks very much.

padeler, 2017-08-01:
The handtracker will only work with NVIDIA GPUs. CUDA is a hard dependency.

closed 2017-08-01 by padeler

KiraRongJin, 2017-08-01:
C:\FORTH\HandTracker>runme.bat
MBV_LIBS Environment variable should point to the MBV libraries fol
MBV_LIBS C:\FORTH\HandTracker
Setting PATH and PYTHONPATH
"Running the Single Hand Tracker script..."
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 8, in <module>
    import PyMBVCore as Core
ImportError: DLL load failed: %1 is not a valid Win32 application.

what is this problem?

KiraRongJin, 2017-08-01:
copy C:\opencv\build\python\2.7\x64\cv2.pyd into python site-package folder...