FORTH-ModelBasedTracker/HandTracker #71: unrecognise option 'libs'

kheito, 2017-08-12:
I connected the 3d hand tracking to labview. It is running and successfully transmit data to labview but the problem is it just suddenly quitting itself after probably few minutes after running it. This is a screenshot of what I am getting when I'm running it. Hope you'll help, thanks in advance
[image]

padeler, 2017-08-16:
Hello,

The warning about the libs environment variable is unrelated to your issues. Can you increase the [log level](https://github.com/FORTH-ModelBasedTracker/HandTracker/blob/master/src/SingleHandTracking.py#L30) to "debug"  and post the log output before the crash? 
Maybe this will give us a clue on whats going on.

kheito, 2017-09-07:
Sorry for the late reply, quick question. How and where can I increase the log severity? 

Thanks

padeler, 2017-09-07:
In the hand tracker script [here](https://github.com/FORTH-ModelBasedTracker/HandTracker/blob/master/src/SingleHandTracking.py#L30) change "error" to "debug"

closed 2017-09-07 by padeler