FORTH-ModelBasedTracker/HandTracker #72: Thumb in roll movement

kimsheena, 2017-09-06:
Hi!

I just observe that when I roll my hand, the 3D Hand model falls on my arms. But when I show my thumb, somewhat the model manages to recover its position. Is that mean that your algorithm is dependent on the thumb?

Kind regards,

padeler, 2017-09-06:
No it is dependent on the structure. 

When the thumb (or fingers in general) are visible on the depth the difference between the palm and the forearm is significant. Thus the objective will have a better score when the model is fitted on the correct position (palm). 

If the fingers are not visible the forearm and the palm are similar as structures in the depth image and the score of the objective is also similar. The result is that the hand will more easily drift to the forearm in the later case.

kimsheena, 2017-09-06:
Hi again!

Thanks for the reply. Just a quick follow-up question. 

So is that mean that to avoid these instances I should just position my hands properly on the figure hand provided before pressing 's' button? Or I should just keep my fingers well separated all through out when running this algorithm??

Kind regards,

padeler, 2017-09-06:
The hand poses that are the easiest for the hand tracker are the ones in which the palm looks different than the forearm. This holds for every frame not just the initialization step. 

kind regards,
--

closed 2017-09-06 by padeler