FORTH-ModelBasedTracker/HandTracker #73: Error running batch script

iDaN5x, 2017-09-08:
Hello,

After installing all the dependencies required, I'm still getting an error when trying to run the batch script:
```cmd
C:\Users\idan5\Desktop\HandTracker>runme.bat
MBV_LIBS Environment variable should point to the MBV libraries folder.
MBV_LIBS C:\Python27\MBV_LIBS
Setting PATH and PYTHONPATH
"Running the Single Hand Tracker script..."
Creating Renderer...
[2017-09-08 13:52:53.376480] [0x00000064] [warning] Read options from environment : unrecognised option 'libs'
[2017-09-08 13:52:53.376480] [0x00000064] [warning] Read options from environment : unrecognised option 'libs'
  Warning! Material file 'C:\Users\idan5\Desktop\HandTracker\media\sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file 'C:\Users\idan5\Desktop\HandTracker\media\cylinder.mtl' not found!
Warning! Material '' not defined in material file.
Variances:  [10.0, 10.0, 10.0, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]
Low Bounds:  [-2000.0, -2000.0, -2000.0, -1.0, -1.0, -1.0, -1.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 2.5, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0]
High Bounds:  [2000.0, 2000.0, 2000.0, 1.0, 1.0, 1.0, 1.0, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.0, 3.140000104904175, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874]
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
Starting Grabber...
[2017-09-08 13:52:54.974963] [0x00000064] [error]   Xn function Failed: The file is corrupted!
[2017-09-08 13:52:54.974963] [0x00000064] [error]   Failed to initialize openni_acquisition plugin. E:\mbv\Acquisition\OpenNIGrabber\OpenNIGrabber.cpp(171): Throw in function void __cdecl MBV::Acquisition::OpenNIGrabber::xnCheckReturnCode(int)
Dynamic exception type: class boost::exception_detail::clone_impl<struct MBV::Core::Exception>
[struct MBV::Core::Message * __ptr64] = OpenNI error. Function Failed

Entering main Loop.
Traceback (most recent call last):
  File "src\SingleHandTracking.py", line 127, in <module>
    c = clbs[0]
IndexError: Index out of range

C:\Users\idan5\Desktop\HandTracker>
```
It might be important to mention that the application's window is opened, but is blank:
[image]

Also, when trying to run NIVIEWER program I'm getting:
```cmd
One or more of the following nodes could not be enumerated:

Device: PrimeSense/SensorKinect/5.1.2.1: The device is not connected!
Device: PrimeSense/SensorV2/5.1.6.6: The device is not connected!
Device: PrimeSense/SensorKinect/5.1.2.1: The device is not connected!
Device: PrimeSense/SensorV2/5.1.6.6: The device is not connected!

Press any key to continue . . .
```

My kinect v2 is connecting and working correctly when using the SDK or Studio.

iDaN5x, 2017-09-08:
@padeler

padeler, 2017-09-08:
Check issue #45 for the openni corrupted oni file. 

Also openni does not work with Kinect v2. That is why your NiViewer cannot open the camera.  If you need to use kinect v2 use the [Kinect2Input branch]( https://github.com/FORTH-ModelBasedTracker/HandTracker/tree/Kinect2Input)

closed 2017-09-08 by padeler

iDaN5x, 2017-09-08:
Switching to Kinect2Input branch worked as described.
Thanks for the help.