FORTH-ModelBasedTracker/HandTracker #74: Get data of the hand pose

zhuzilin, 2017-09-08:
Hi! Nice work! Thank you for providing such great demo. However, I wonder if there is any interface from which I could extract the specific 3D data of the hand? For example the angle and Cartesian coordinates. Thank you.

zhuzilin, 2017-09-08:
Also, is there a way to change the parameter of the hand model? Thank you.

padeler, 2017-09-11:
Regarding your first question check #16  

For the second, unfortunately the current hand model is described with an xml file which is kind of cumbersome to edit. For example you can change the scale of the hand from [here](https://github.com/FORTH-ModelBasedTracker/HandTracker/issues/16)

In a future release we might change this to a more general model format (i.e blender files).

closed 2017-09-11 by padeler