FORTH-ModelBasedTracker/HandTracker #76: With and without PSO data

kimsheena, 2017-10-02:
Hi!

Is it possible to get a data of your algorithm using hypothesis scoring only (w/o pso)? Our study is about application of your algo in mimicking a robotic hand, and we are ask about the comparison of having a pso and without it. We conclude that data will help in convincing them. 

Does your study provide a data w/ and w/o using pso? Or if not, how can we decode it and eliminate pso in the process for the meantime? Please do help

Thanks

padeler, 2017-10-04:
Hello,

You can use step6_evaluate instead of step6_track. 

**step6_evaluate** is overloaded on the arguments:

If you provide a Core.ParamVector with a single pose it will return a tuple of the following:
> obsDepth, obsLabel, hypDepth, hypIssue, score, conjunction, disjunction
From the above you need the **score**.

If you provide a Core.ParamVectors which is a vector of poses it will evaluate all given poses and will return a list with the scores, one for each evaluated pose.

Alternatively you can run PSO with particles=generations=1 and it will just return the score for the initial pose.

I hope it helps.

kimsheena, 2017-10-07:
Hi again!

I tried replacing step6_track to step6_evaluate but when I press the 's' button, the handtracking crash and close. 

Thanks by the way for the response, very much appreciated!

padeler, 2017-10-16:
Hello,

Do you get any messages when it crashes? can you give a little more information? Of-course there will be no tracking when evaluating a single hypotheses against the observation.
 
Did you try with particles=generations=1 and got the same result? This option should work since you already have a working tracker.

closed 2018-02-28 by padeler