FORTH-ModelBasedTracker/HandTracker #79: Is this project compatible with DS325 camera?

WuXinyang2012, 2018-02-27:
Hi, I have one SoftKinetic DS325 camera and try to adapt your amazing work on it, however I am not sure where to start it.

padeler, 2018-02-28:
You need to replace the code of the grabber with your own.  Check the code examples in the lines were the OpenNIGrabber is used to grab frames.

You can use any depth sensor you want the tracker just needs a depth frame and the intrinsic camera parameters (calibration). The rgb frame is only used for visualization so if you do not have it you can just provide a dummy image.

Ethanwl, 2018-03-13:
If I use Binocular camera to generate a depth frame, what should I do to sent it to the program? What 
code should I change when I use Binocular camera?

padeler, 2018-03-23:
Depth from stereo is not good enough for hand tracking using this method. 

That said, you can test the performance of the handtracker with stereo input by using the depth image as well as one of the two RGB images (i.e left rgb). 
As with all other cameras you have to set the intrinsic and extrinsic parameters of the camera in the "calib" object.