FORTH-ModelBasedTracker/HandTracker #80: RuntimeError at OpenGLHelpers.cpp(xxx)

vickailiu, 2018-04-24:
Hi,

I'm trying to test this project on Ubuntu 16, Cuda 9.1, for realsense camera.

I rebuilt opencv as mentioned in and put them in the libs and resolved opencv problem (https://github.com/FORTH-ModelBasedTracker/HandTracker/issues/2).
I followed https://github.com/FORTH-ModelBasedTracker/HandTracker/issues/48 for realsense camera.

Currently I encountered this problem during runtime (see the log at the bottom). The last chunk is repeating. I verified that my cuda is working as mentioned in https://github.com/FORTH-ModelBasedTracker/HandTracker/issues/4. 

May I know how I can further investigate and fix it?

```
 [2018-04-24 10:18:38.606462] [0x000007f4eed30f70] [info]    Max color attachments = 8
[2018-04-24 10:19:24.698096] [0x000007f4eed30f70] [info]    Grid product initialized with size = 32 X 32
[2018-04-24 10:19:24.699139] [0x000007f4eed30f70] [info]    Creating default decoder
  Warning! Material file '/home/lk/experimental/media/sphere.mtl' not found!
Warning! Material '' not defined in material file.
  Warning! Material file '/home/lk/experimental/media/cylinder.mtl' not found!
Warning! Material '' not defined in material file.
[2018-04-24 10:19:24.775434] [0x000007f4eed30f70] [info]    loadMeshTickets
[2018-04-24 10:19:24.775448] [0x000007f4eed30f70] [info]    Nodes : 55
[2018-04-24 10:19:24.775452] [0x000007f4eed30f70] [info]    Stages : 5
[2018-04-24 10:19:24.775456] [0x000007f4eed30f70] [info]    Stage 0 : 1 multiplications
[2018-04-24 10:19:24.775459] [0x000007f4eed30f70] [info]    Tickets :
[2018-04-24 10:19:24.775462] [0x000007f4eed30f70] [info]    	18446744073709551615
[2018-04-24 10:19:24.775466] [0x000007f4eed30f70] [info]    Stage 1 : 8 multiplications
[2018-04-24 10:19:24.775470] [0x000007f4eed30f70] [info]    Tickets :
[2018-04-24 10:19:24.775474] [0x000007f4eed30f70] [info]    	0
[2018-04-24 10:19:24.775486] [0x000007f4eed30f70] [info]    	1
[2018-04-24 10:19:24.775491] [0x000007f4eed30f70] [info]    	18446744073709551615
[2018-04-24 10:19:24.775495] [0x000007f4eed30f70] [info]    Stage 2 : 15 multiplications
[2018-04-24 10:19:24.775499] [0x000007f4eed30f70] [info]    Tickets :
[2018-04-24 10:19:24.775504] [0x000007f4eed30f70] [info]    	0
[2018-04-24 10:19:24.775508] [0x000007f4eed30f70] [info]    	1
[2018-04-24 10:19:24.775512] [0x000007f4eed30f70] [info]    	18446744073709551615
[2018-04-24 10:19:24.775518] [0x000007f4eed30f70] [info]    Stage 3 : 15 multiplications
[2018-04-24 10:19:24.775523] [0x000007f4eed30f70] [info]    Tickets :
[2018-04-24 10:19:24.775529] [0x000007f4eed30f70] [info]    	0
[2018-04-24 10:19:24.775533] [0x000007f4eed30f70] [info]    	1
[2018-04-24 10:19:24.775536] [0x000007f4eed30f70] [info]    	18446744073709551615
[2018-04-24 10:19:24.775540] [0x000007f4eed30f70] [info]    Stage 4 : 15 multiplications
[2018-04-24 10:19:24.775544] [0x000007f4eed30f70] [info]    Tickets :
[2018-04-24 10:19:24.775547] [0x000007f4eed30f70] [info]    	0
[2018-04-24 10:19:24.775551] [0x000007f4eed30f70] [info]    	1
[2018-04-24 10:19:24.775558] [0x000007f4eed30f70] [info]    Creating default BG/FG
[2018-04-24 10:19:24.775563] [0x000007f4eed30f70] [info]    Setting default configuration
[2018-04-24 10:19:24.775567] [0x000007f4eed30f70] [info]    Configuring objective function
Variances:  [10.0, 10.0, 10.0, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]
Low Bounds:  [-2000.0, -2000.0, -2000.0, -1.0, -1.0, -1.0, -1.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 2.5, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0, 0.0, 1.2999999523162842, 0.0, 0.0]
High Bounds:  [2000.0, 2000.0, 2000.0, 1.0, 1.0, 1.0, 1.0, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.0, 3.140000104904175, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874, 1.5700000524520874, 1.840000033378601, 1.5700000524520874, 1.5700000524520874]
Randomization Indices:  [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
[ERROR] [1524536364.977708]: bad callback: <bound method image_converter.depthCallback of <__main__.image_converter instance at 0x7f4eb4793128>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 161, in depthCallback
    self.process()
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 192, in process
    bb = self.ht.step2_computeBoundingBox(self.currentHandPose, width, height, 0.1)
RuntimeError: /home/padeler/work/mbv/release_libs/src/mbv/Rendering/RendererOGL/OpenGLHelpers.cpp(370): Throw in function void OGL::FrameBufferObject::check()
Dynamic exception type: boost::exception_detail::clone_impl<MBV::Core::Exception>
[MBV::Core::Message*] = no error

[ERROR] [1524536364.979753]: bad callback: <bound method image_converter.depthCallback of <__main__.image_converter instance at 0x7f4eb4793128>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 161, in depthCallback
    self.process()
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 192, in process
    bb = self.ht.step2_computeBoundingBox(self.currentHandPose, width, height, 0.1)
RuntimeError: /home/padeler/work/mbv/release_libs/src/mbv/Rendering/RendererOGL/RendererOGL.cpp(684): Throw in function virtual void MBV::Rendering::RendererOGLBase::beginBatch(MBV::Rendering::RenderingFlags)
Dynamic exception type: MBV::Core::Exception
[MBV::Core::OffendingCode*] = m_data->flags == 0
[MBV::Core::Message*] = already in batch

[ERROR] [1524536364.981886]: bad callback: <bound method image_converter.depthCallback of <__main__.image_converter instance at 0x7f4eb4793128>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 161, in depthCallback
    self.process()
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 192, in process
    bb = self.ht.step2_computeBoundingBox(self.currentHandPose, width, height, 0.1)
RuntimeError: /home/padeler/work/mbv/release_libs/src/mbv/Rendering/RendererOGL/RendererOGL.cpp(684): Throw in function virtual void MBV::Rendering::RendererOGLBase::beginBatch(MBV::Rendering::RenderingFlags)
Dynamic exception type: MBV::Core::Exception
[MBV::Core::OffendingCode*] = m_data->flags == 0
[MBV::Core::Message*] = already in batch

[ERROR] [1524536364.984483]: bad callback: <bound method image_converter.depthCallback of <__main__.image_converter instance at 0x7f4eb4793128>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 161, in depthCallback
    self.process()
  File "/home/lk/experimental/src/emma_commons/scripts/listener.py", line 192, in process
    bb = self.ht.step2_computeBoundingBox(self.currentHandPose, width, height, 0.1)
RuntimeError: /home/padeler/work/mbv/release_libs/src/mbv/Rendering/RendererOGL/RendererOGL.cpp(684): Throw in function virtual void MBV::Rendering::RendererOGLBase::beginBatch(MBV::Rendering::RenderingFlags)
Dynamic exception type: MBV::Core::Exception
[MBV::Core::OffendingCode*] = m_data->flags == 0
[MBV::Core::Message*] = already in batch

```

padeler, 2018-05-03:
This issue looks like it is not related to the handtracker but in the way the method is called from your scripts. 

Do you make multiple calls from different threads to the handtracker instance? (the handtracker lib is not designed to be thread safe for use by multiple threads)

Do the handtracker example scripts work correctly?

vickailiu, 2018-05-04:
@padeler shouldn't be, but I'll double confirm.

vickailiu, 2018-05-07:
hi @padeler I'm trying the example scripts with static image. i.e. I read the previous stored rgb and depth image. May I know how can I get the depth image format correct? currently I'm having the error showing
```
[MBV::Core::Message*] = Depth part muse be ushort
```

lj-cug, 2021-12-29:
I have the same problem. do you have solved it?