FORTH-ModelBasedTracker/HandTracker #82: import error about 'libcore.so'

geoffzhang, 2018-07-09:
mportError: /usr/lib/handtrack/libs/libCore.so: undefined symbol: _ZN2cv9ExceptionC1EiRKSsS2_S2_i
please, how should I modifiy this error?

maohuui, 2019-02-12:
i also encountered the same problem ,how do you solve it?
i set the environment variable.

padeler, 2019-02-12:
This is probably a opencv version issue. These libraries where built for ubuntu 14.04 with the opencv v2.48.

maohuui, 2019-02-12:
i see. thank you

jiazaizhong52, 2021-04-05:
I encountered the same problem too. I found that the problem is possibly caused by the wrong version of gcc(& g++). the same error was reported when I used gcc-5. Finally when I changed to gcc-4 and rebuild the opencv package, this problem was solved.