FORTH-ModelBasedTracker/HandTracker #83: Incomatible with today's software

kalsan, 2018-10-31:
Ubuntu 14.04, the version this HandTracker appears to be made for, is going to be end-of-life within less than a year (https://www.ubuntu.com/about/release-cycle). The library `libcore.so` does not work on Ubuntu 18.04 since it's linked against `libopencv_core.so.2.4` but today's systems have `libopencv-core3.2`. Installing 2.4 manually is not enough to make HandTracker operational.

Due to this massive leap in time HandTracker risks to be end-of-life within a year or two if it is not updated. Please consider providing a new version of the FORTH libraries that works with up-to-date systems.

padeler, 2018-10-31:
You are right, a new release of the hand tracker binaries (with additional features) is something we plan to do but always gets delayed by other more pressing requirements.

use-my-life, 2023-07-05:
oh, this project is amazing, so i want to repeat this project. now, could you update this project? thank you